ros-drivers/velodyne #326: Fix vert_correction is miss-naming in yaml file

jerry73204, 2020-01-17:
For example the [VLP-32C parameters yaml](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/params/VeloView-VLP-32C.yaml), the _rot\_correction_ is the offset added to the azimuth, however, _vert\_correction_ is the angle elevating from the xy plane. That is confusing to me at first. I think we may adopt the name _elevation angle_ in VLP-32C User Manual 63-9325 Rev. C manual, which is more clear and precise.

JWhitleyAStuff, 2020-01-22:
PRs are welcome!

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): The requested terminology change has not landed: ROS1 `master`, `melodic-devel`, and ROS2 `ros2` still parse the YAML key `vert_correction`, and the VLP-32C calibration file uses that key. Merged [#518](https://github.com/ros-drivers/velodyne/pull/518) updates VLP-16 vertical-offset values but does not rename the field. This remains a naming/schema request; I did not find evidence that the numeric correction itself is wrong.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/src/lib/calibration.cc#L39-L65).