ros-drivers/velodyne #327: ROS2 Transform Node Memory Error

Chase-san, 2020-01-30:
**Please complete the following information:**
 - OS and Version: Ubuntu 18.04 Docker
 - ROS Version: ROS2 Dashing
 - Built from Source or Downloaded from Official Repository: Built from Source
 - Version: 2cbd8a47f23b9ef877d0ccb34e410499b451c77f (dashing-devel head)

**Describe the bug**
velodyne/velodyne_pointcloud/src/conversions/transform.cpp  Line 161

Here a new Shared Pointer is created, however this appears to cause a memory error.

**To Reproduce**
1. Configure a velodyne driver normally
2. Configure transform node to have the same frame_id and fixed_frame as the DRIVER frame_id.
3. Power on velodyne.
4. Launch the driver and transform node with composed launch.
5. Transform node crashes with "free(): invalid size" error.

**Expected behavior**
The transform node does not crash, and filters based on frame_id.

**Additional context**
I fixed this by having a "const shared_ptr<const VelodyneScan>" for setup in both the data container base and the point cloud classes, which prevents the need to create the shared pointer, however this leads to significant code duplication.

I ran into this issue as I am running both a VLP32C and a VLP16 on the same ros2, which causes conflicting conversions to occur, as the VLP32C convert node was converting VLP16 data, as well as it's own and vice versa. As the topics are not configurable, I found the transform node allows filtering based on frame id.

JWhitleyAStuff, 2020-01-31:
Thanks for reporting the issue, @Chase-san! Would you mind creating a Pull Request to resolve this issue?

clalancette, 2020-07-10:
> I ran into this issue as I am running both a VLP32C and a VLP16 on the same ros2, which causes conflicting conversions to occur, as the VLP32C convert node was converting VLP16 data, as well as it's [...]

I will note that I've tested both a VLP16 and VLP32C with dashing-devel when I did the initial port here.  Both worked fine for me.

But I think this may be a mis-configuration.  Topics are *always* configurable in ROS 2; you just need to use the topic remapping, like:

```
ros2 run velodyne_driver velodyne_driver_node --ros-args -r velodyne_packets:=/vlp16/velodyne_packets
```

What definitely will *not* work is having two velodyne_drivers from two different VLPs publishing to the same `/velodyne_packets` topic.  That will cause them to interfere with each other, and may be the reason you are getting a crash here.

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): On the reported `dashing-devel` branch, `processScan()` still casts away const and creates a new mutable `shared_ptr` from the scan pointer before calling `setup()`; `setup()` accepts that mutable pointer. The current ROS2 `ros2` branch instead passes `ConstSharedPtr` through the setup path without that cast. A maintainer also noted that separate drivers should use topic remapping rather than publish to the same packets topic; that addresses the multi-driver configuration concern but not the pointer construction above. I did not run a Dashing reproduction or establish a fix on that reporting branch, so the issue remains open.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/e154fd275f52ef5cae60a4a78f31ec5582afbc11/velodyne_pointcloud/src/conversions/transform.cpp#L150-L165).