ros-drivers/velodyne #332: Use use_sim_time param to avoid running driver when only replay is desired

kaarta-SHanna, 2020-02-18:
**Is your feature request related to a problem? Please describe.**
If I want to run a replay but still have my sensor connected, the driver still runs and will produce packets.

**Describe the solution you'd like**
Add an if statement to query the /use_sim_time parameter. It is a standard ROS param and thus should only ever be set to true when replaying data, so I don't see this causing any issues. If it's set to true, don't run the driver component.

**Describe alternatives you've considered**
Most other alternatives require a remapping of topics, turning off network interfaces, disconnecting the sensor, etc. None are super simple

kaarta-SHanna, 2020-02-18:
I'm happy to implement this, but would like other feedback to see if it's desired by the community

JWhitleyWork, 2020-02-19:
When you say "replay," do you mean rosbag replay or pcap replay?

kaarta-SHanna, 2020-02-19:
Replay meaning rosbag replay, since that is causing the issues where we are playing bag data and live sensor data at the same time

JWhitleyWork, 2020-02-22:
@kaarta-SHanna - This isn't an issue of something that needs to be fixed in the driver, it's just something that needs to be changed in your launch file. If you are using the default launch file for a given sensor, it will always launch the driver. If your bag file has both `/velodyne_points` and `/velodyne_packets`, then you don't need to run any part of the `velodyne` package when you're replaying a bag file. If your bag file only has `/velodyne_packets`, then you need to run the `velodyne_pointcloud` node without the driver, for which you'll have to create a custom launch file.

closed 2020-02-22 by JWhitleyWork