ros-drivers/velodyne #334: Has anyone tested 'dashing-devel` on a Velodyne HDL 64 lidar?

fmg-shehan, 2020-03-17:
**This is not a bug but thought I'd ask this question under this template**

**Compute setup:**
 - OS and Version: Ubuntu 18.04
 - ROS Version: dashing
 - Built from: Built from source
 - Version: Version from 'dashing-devel' branch

**Describe the bug**
Trying to test the dashing-devel version on a HDL 64 lidar but keen to know if someone has already tested this?

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repo in a catkin ws.
2. Built with colcon.

**Expected behavior**
Read through the HDL 64 lidar packets and publish a `points2` topic.

fmg-shehan, 2020-03-17:
@JWhitleyWork tagging you for your attention. Not sure if you're the right person ask the question though. Thanks

JWhitleyWork, 2020-03-20:
I have not personally tested this. I do not know if anyone else has. I'll leave the question open for a bit to see if anyone else responds.

fmg-shehan, 2021-08-03:
Closing as there were no response

closed 2021-08-03 by fmg-shehan