ros-drivers/velodyne #336: Customize output topic name

rongguodong, 2020-03-26:
It seems the current output topic name is hard-coded as "velodyne_points". Is it possible to customize this name in config/launch files, without modifying the source code?

JWhitleyWork, 2020-03-27:
Please see [this page](http://wiki.ros.org/roslaunch/XML/remap) on topic remapping.

closed 2020-03-27 by JWhitleyWork

hadiTab, 2020-03-27:
@JWhitleyWork does making it configurable cause any issues? i.e. if we make the topic name a configurable param that can be set in the launch file with a default topic name and make a pull request would there be any issues with merging it?

JWhitleyWork, 2020-03-28:
@hadiTab While this is technically possible, this is a complex topic. Some community members feel that it is a good way to "get something to work" while others say it should almost never happen. See the discussion on [this ROS Answers question](https://answers.ros.org/question/342777/is-it-correct-to-set-topic-names-from-parameters/). In the end, I consider the Velodyne driver to have a pretty static API and, as such, would fall into the category of "makes more sense to remap." In addition, the API has stood for quite some time so changing it for convenience for a single use-case doesn't seem like the best idea.