ros-drivers/velodyne #341: Package in ROS offical repository(APT) is out of date

flymaxty, 2020-05-03:
Hi,
I'm trying to use velodyne-pointcloud with param "organize_cloud", and I found that the package which installed by apt from ros offical repository, is different from master branch. even though they are both version 1.5.2(veldoyne_pointcloud).

I don't know how apt repository works, but looks like it is caused by non-update version number.
Is there any plan to make a new release for this repo?

Please let me know, if I have something misunderstood.
Thank you!

JWhitleyWork, 2020-05-03:
@flymaxty This is correct. I have not released a new version lately because of problem described in #340. Once this is resolved, I will release a new version.

closed 2020-05-03 by JWhitleyWork

flymaxty, 2020-05-03:
Thanks for your reply! Looking forward to new release