ros-drivers/velodyne #342: Confusing naming of TF frame parameters in velodyne_pointcloud package

AndreasR30, 2020-05-20:
You use the following parameter names to specify TFs:
1. **frame_id**: "fixed frame of reference for point clouds" (According to TransformNode.cfg)
  1.1. Is stored in the code to _target_frame_ in reconfigure callback:
`config_.target_frame = tf::resolve(tf_prefix_, config.frame_id);`
  1.2. So why it has a different name in dynamic reconfigure?
2. **fixed_frame**: "The desired input frame" (According to CloudNode.cfg)
  2.1. Does not seem to be used anywhere
  2.2. Input frame should be clear from the header of the _velodyne_packets_ message
  2.3. So there should be no need to specify the input frame explicitly, right?
3. **target_frame**: "The desired output frame" (According to CloudNode.cfg)
  3.1. This is the frame used for inertial correction of the velodyne packets in Transform Node:
`tf_ptr->lookupTransform(config_.target_frame, cloud.header.frame_id, time, transform);`
  3.2. So for the Transform Node it is typically set to something like "odom"
  3.2. For both, Transform and Cloud Node: This is also the output frame in the header of the final _velodyne_points_ message

I would propose the following naming convention:
1. **frame_id**: **Just** the input frame for the **velodyne_driver**
  1.1. This frame is stored into the header of the _velodyne_packets_ message
2. **fixed_frame**: This should be the reference frame used for inertial correction
  2.1. As it is **fixed** in the world frame (required to perform inertial correction)
  2.2. Consequently set to something like "odom"
3. **target_frame**: This is the target/ouput frame of the final _velodyne_points_ message
  3.1. Nice feature to specify a frame to which the pointcloud (PC) should be finally transformed
  3.2. For example to transform the PC back to velodyne frame after performing inertial correction (currently the oulput frame would be forcibly something like "odom")

AndreasR30, 2020-05-20:
Maybe even better (more intuitive) names for the TFs would be (given my proposals):
- **frame_id** -> **sensor_frame**
- **fixed_frame** -> **odom_frame** (only required for Transform Node)
- **target_frame** -> **target_frame** or ouput_frame

Maybe another reason to use the new names would be to still support the old behavior (for backwards compatibility) while at the same time supporting the proposed behavior with the new names

JWhitleyWork, 2020-05-21:
@AndreasR30 I very much agree with this but don't have time to create a Pull Request to apply the change. Could you please create one and I will review?

JWhitleyWork, 2020-05-21:
To answer your questions from above, I'll give an example of what I "think" the original author was going for:

`frame_id`: The `frame_id` parameter is passed down to the `velodyne_driver` which produces the `velodyne_packets` topic. It is used there to populate the `frame_id` field in the header of the messages which are published to `velodyne_packets`. `sensor_frame` is fine as a replacement. `packets_frame` would also work but I don't know if you feel this one is confusing or not.

`fixed_frame`: The `fixed_frame` parameter is passed down to the `CloudNodelet` which holds an instance of a `Convert` object (`src/conversions/convert.cc`). The `Convert` object then passes it to the `PointcloudXYZIR` or `OrganizedCloudXYZIR` object (depending on whether `organize_cloud` is true or not). This is then passed to the parent class, `DataContainerBase`. In this class, all it's used for is to compare it against `target_frame` and see if they are the same. If they are not, it creates a `tf::TransformListener`. For some reason, the `computeTransform` function then uses the `TransformListener` to look up a transform between the `target_frame` (**not** the `fixed_frame`) and the `frame_id`  that is populated in the cloud message, which, in the `setup` function, is set to the same frame as what's received in the `velodyne_msgs::VelodyneScan` (the raw packets in the `velodyne_packets` topic). Utterly stupid. This can be gotten rid of completely and `computeTransformation` can be modified to see if `target_frame` and the `frame_id` from the scan are different - if they are, then look up _that_ transform.

`target_frame`: This does its job correctly. As the name implies, it's the frame that you want the `velodyne_points` to be published in, after a transform, if one is necessary. I'm fine with this one as-is.

AndreasR30, 2020-05-23:
I am happy that you also see the problems and Thank You for your detailed answer. I will make a pull request the next days.
Before this, I have some comments to your answer:
The `CloudNodelet` does not do any TF transformations as `computeTransform` is never called in `convert.cc` (when I see it correctly). In fact it does not make sense to specify **any** TF frame for it because not a single TF is perfomed. Very strange. I'm not sure whether this is intended or just a bug.
In general I would completely get rid of `convert.cc` as it is almost the same as `transform.cc` with less functionality. I think it would be good to avoid this code duplication as it is error-prone. So I would merge both into one.

In general I see the following 4 scenarios:

1. Someone just wants to convert the `VelodyneScan` to a `PointCloud2` message, which should be in the same sensor frame:
  1.1. In current version: done with `convert.cc` / `CloudNodelet`, which asks for unnecessary TF frames --> very confusing
  1.2. In new version: no TF frame has to be specified as the frame can be obtained from `VelodyneScan` msg. So when no frame is specified this behavior is triggered.
2. Same as _1._ but the output frame of the PC should be different to the sensor frame:
  2.1. In current version: I think this would require to use `transform.cc` / `TransformNodelet`, which (I think) was originally designed for inertial correction (ego motion compensation) and not for this scenario
  2.2. In new version: When `target_frame` is specified then the whole PointCloud is transformed to the desired frame
3. Someone wants a inertial corrected `PointCloud2` message while the output frame of the resulting PC is irrelevant. This is different to `2.` as it requires to use different transforms for each package (theoretically even for each point) as they have different timestamps which correspond to different positions of the ego vehicle. In _2._ the whole PC can be transformed with a single homogeneous transform.
  3.1. In current version: This is achieved with `transform.cc` / `TransformNodelet`. Works well, however unclear roles of `fixed_frame` & `target_frame`
  3.2. In new version: This behaviour is triggered when just `fixed_frame` (the odom frame) is specified.
4. Same as 3. but the inertial corrected PC should be in a specified target frame (other than odom). Typically one wants to have the PC back in the sensor frame as this is important for downstream segmentation steps.
  4.1. In current version: Not possible (even though the TF names suggest this)
  4.2. In new version: This behavior is triggered when `fixed_frame` **and** `target_frame` are specified. First each Velodyne packet is transformed individually to `fixed_frame` (the odom frame) based on its timestamp in order to take the ego motion into account. Afterwards the whole PC is transformed as a whole to the desired  `target_frame` (again a single transform for whole PC)

spuetz, 2020-06-09:
Hmm normally the `fixed_frame` is the world or map frame, which is the root of the tf (sub-)tree containing the target frame and the sensor_frame.

UniBwTAS, 2020-06-09:
> Hmm normally the `fixed_frame` is the world or map frame, which is the root of the tf (sub-)tree containing the target frame and the sensor_frame.
> See also: http://docs.ros.org.ros.informatik.uni-freiburg.de/jade/api/tf2/html/classtf2_1_1BufferCore.html#a362de8ea45b99efde65bcc602053ad38

Yes in the Pull Request above this is now the case. One can specify map as the fixed_frame. Then ego motion compensation is performed. However, it is better to specify odom instead of map because there are no discrete jumps of sensor_frame within odom (unlike map or world frame). See:
https://www.ros.org/reps/rep-0105.html#odom

spuetz, 2020-06-09:
Japp japp, I know that it is better to use odom is such a use case. ;)

closed 2020-07-09 by JWhitleyWork