ros-drivers/velodyne #345: Add .clang-format

AndreasR30, 2020-05-26:
The format of the source code is quite different to the one of my IDE. Furthermore it is not really consistent within this repository. During my last pull request, I lost a lot of time manually following the existing format. Therefore I would suggest to add a .clang-format file to the root of this repository. There is a official [ROS CppStyleGuide](http://wiki.ros.org/CppStyleGuide#Autoformatting_of_ROS_Code), which links to a .clang-format file.

JWhitleyWork, 2020-05-26:
@AndreasR30 I'm fine with updating the code to match the style guide (`roslint` is already applied but this doesn't catch all of the things that Clang does) but I think adding the file to the repo might be excessive. If you want to submit MRs to make the repo comply with https://github.com/davetcoleman/roscpp_code_format/blob/master/.clang-format, I'm happy to accept them.

YoshuaNava, 2020-09-10:
@AndreasR30 Do you think this could be addressed? +1

@JWhitleyWork Keeping the clang-format file I think is actually a good thing, as people can trigger formatting from the IDE (vscode, clion, sublime), and it also makes the repo hold a copy of its own format, in case the external repo is deleted. In any case, we could add a header to the copy of the format file saying that it is a copy of another source and it is kept there for reference. What do you think?

JWhitleyWork, 2020-09-11:
I'd be OK with that.