ros-drivers/velodyne #350: time field meaning

narutojxl, 2020-07-07:
Hi authors,
I found the updated velodyne driver can output `time` field. When i transform the ros msg format `sensor_msgs::PointCloud2` to PCL format `pcl::PointCloud<PointXYZIRT>` with `pcl::fromROSMsg(const sensor_msgs::PointCloud2 &cloud, pcl::PointCloud<T> &pcl_cloud)` API, the every point has a time. 
What is the time meaning? Is the point's timestamp minus the got current PointCloud2 ros msg timestamp? BTW, whose point timestamp is the timestamp of a received ros msg. I am useing velodyne-VLP-16, with `roslaunch velodyne_pointcloud VLP16_points.launch` to get data.
Thanks for your help!

JWhitleyWork, 2020-07-09:
Your wording is confusing so I will give some general information about the timestamps. Each point has a timestamp which is calculated based on the sensor's timestamp and the firing sequence. However, whether the timestamp was generated by the sensor or is an estimate by the driver depends on whether or not you are using the `use_gps_time` parameter. See the code at https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/lib/input.cc#L226 for more details.

As for the message timestamp, that depends on the parameter `timestamp_first_packet`. If this is true, then the message timestamp is the same as the timestamp of the first packet. If it is false, then the message timestamp is that of the last packet.

narutojxl, 2020-07-10:
Hi @JWhitleyWork, thanks for your reply and time.
Sorry, i didn't describe my problem clearly,
> 
    pointCloud_callback(const sensor_msgs::PointCloud2ConstPtr &laserCloudMsg) {
      sensor_msgs::PointCloud2 currentCloudMsg = *laserCloudMsg;
      double msg_timestamp = currentCloudMsg.header.stamp.toSec();

      pcl::PointCloud<PointXYZIRT>::Ptr laserCloudIn;
      pcl::fromROSMsg(currentCloudMsg, *laserCloudIn);
      int cloudSize = laserCloudIn->points.size();
      for (int i = 0; i < cloudSize; i +) {
        pcl::PointXYZI p;
        p.x = laserCloudIn->points[i].x;
        p.y = laserCloudIn->points[i].y;
        p.z = laserCloudIn->points[i].z;
        p.intensity = laserCloudIn->points[i].intensity;
        int ring = laserCloudIn->points[i].ring;
        double time = laserCloudIn->points[i].time
      }
    }

I don't really understand the driver code totally.
* For msg_timestamp,  when without GPS,  is `timestamp_first_packet` equal the first point's timestamp in the scan? I am not familiar with  the word `packet`.
* For each point in the scan, when without GPS, `double time = laserCloudIn->points[i].time`:   `time`  variable equals  `msg_timestamp - point's time`  or  directly equals  `point's time` ?
Thanks for your kind help a lot !

JWhitleyWork, 2020-07-17:
Answer to first question:
- When `timestamp_first_packet` = true, the *message* timestamp is equal to first point's timestamp in scan.
- When `timestamp_first_packet` = false, the *message* timestamp is equal to last point's timestamp in scan.

Answer to second question:
- See code at https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/lib/input.cc#L226

narutojxl, 2020-07-17:
Thanks a lot @JWhitleyWork !

closed 2020-07-17 by narutojxl

alexsaso, 2021-04-21:
I would like to ask again one of @narutojxl questions because I did not find it answered here. When I look at one lidar message, it contains some information for 24x16 points, so every single ray was fired 24 times. Each of these points now contains 6 values, the x, y, z - coordinates, intensity, ring and time.

Adding to @narutojxl question, I would like to know how I have to read the time field of each point. Is it as he suggested the negative offset from the messages timestamp or rather a positive offset? So that a points time is calculated as point.time_corrected = msg.timestamp + point.time, or rather as point.time_corrected = msg.timestamp - point.time? I could not find this part in the code linked by @JWhitleyWork.

By time_corrected I mean the actual time when that point was created. msg.timestamp refers to the the timestamp of the packet, being the time of the firing of the first or the last firing in that message, as defined by timestamp_first_packet. Lastly point.time is the value contained in the timefield of that point.

Another thing which I noticed was, that right now most of the timefields in the messages which I receive using this driver are set to zero. Only the fields of the points coming from ring 0 actually have a positive value corresponding to the time it takes for one firing sequence to complete, as defined in the Velodyne manual. So the values are 3.456e-5, 8.9856e-5, 1.45152e-4 and so on. Is this the normal behavior for the driver?