ros-drivers/velodyne #353: Release dashing-devel branch into Dashing, Eloquent, and Foxy

clalancette, 2020-07-10:
@JWhitleyWork Opening this up to see if you'd like to do a binary release of this driver into Dashing, Eloquent, and Foxy.  I think this has been around long enough that it would be worthwhile to make this `apt-get` installable.  Thoughts?

JWhitleyWork, 2020-07-10:
Hey, @clalancette. The `dashing-devel` branch currently targets Dashing and the `ros2` branch targets Foxy. I'm testing the `ros2` branch now for compatibility with Eloquent and adding CI tests for it. I'd be happy to release for Dashing to get started, though.

clalancette, 2020-07-10:
> Hey, @clalancette. The `dashing-devel` branch currently targets Dashing and the `ros2` branch targets Foxy. I'm testing the `ros2` branch now for compatibility with Eloquent and adding CI tests for it. I'd be happy to release for Dashing to get started, though.

For what it is worth, we've been using the `dashing-devel` branch on Eloquent for quite a while.  So I'm fairly confident it works there.

I haven't really tested the `ros2` branch on Foxy, but it does at least compile there :).

JWhitleyWork, 2020-07-10:
@clalancette Do you think the APIs between Dashing and Eloquent are more similar or between Eloquent and Foxy? I just added Eloquent CI to the `ros2` (Foxy) branch and it passes without error or warning.

JWhitleyWork, 2020-07-10:
Dashing PR: https://github.com/ros/rosdistro/pull/25756

clalancette, 2020-07-10:
> @clalancette Do you think the APIs between Dashing and Eloquent are more similar or between Eloquent and Foxy?

I think Dashing and Eloquent are slightly closer together than Eloquent and Foxy.  But for the purposes of this package as it stands today, there are no major API differences between all three of them.

JWhitleyWork, 2020-07-10:
I think I'd rather set a precedent of maintaining compatibility with older non-LTS releases until they EOL than trying to attach compatibility to newer releases to existing branches (have `ros2` support `eloquent` and `foxy` rather than `dashing-devel` support `eloquent`). This way, when `rolling` becomes the default, we'll cut a `foxy-devel` branch which will support Foxy and Eloquent. Though I guess it won't matter much once `rolling` is a thing. Anyway, I'll release `ros2` in both Eloquent and Foxy for now.

clalancette, 2020-07-10:
> Anyway, I'll release `ros2` in both Eloquent and Foxy for now.

Awesome, thanks!  It's appreciated.

JWhitleyWork, 2020-07-10:
Eloquent PR: https://github.com/ros/rosdistro/pull/25758
Foxy PR: https://github.com/ros/rosdistro/pull/25759

closed 2020-07-10 by JWhitleyWork