ros-drivers/velodyne #356: Velodyne Driver Catkin build on Windows 10 for  LG Simulator Lidar Plugin

kilocharlievictor, 2020-07-19:
Hi,
We have been trying to build the source for the Lidar Plugin integration in LG Simulator. 
Following link points to this git clone page: https://www.lgsimulator.com/docs/lidar-plugins.
 Having installed ROS for Windows from this page: http://wiki.ros.org/Installation/Windows,  I am trying to build the source on WIndows machine. 
Fails to build with following failure. Is this source meant for building on Windows at all?
Please advise..
Regards
[image]
[image]

[image]

JWhitleyWork, 2020-07-19:
Which branch are you using? I do not believe it has been tested on Windows. I know there are parts of the architecture (the UDP interface) that are Linux-specific but they can be replaced with something like [udp_driver](https://github.com/ros-drivers/transport_drivers).

kilocharlievictor, 2020-07-22:
Hi,
I used the master. 
catkin_make fails.

JWhitleyWork, 2020-07-29:
@kilocharlievictor Unfortunately, I need significantly more information than this to provide any help. I also do not have a Windows machine on which to test so I don't know how much help I can be. It is unlikely that I will be able to provide support for ROS1 on Windows so you will probably have to troubleshoot yourself. If you have any specific questions, I'll be as much help as I can. I'll leave this ticket open in case there is more discussion.

kinger-ml, 2020-10-01:
@JWhitleyWork I also tried installing the package in windows.
On performing `rosdep install`, i got the following error:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
velodyne_driver: No definition of [libpcap] for OS [windows]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully`

Is there anything I can try to resolve this?

JWhitleyWork, 2020-12-03:
@kinger-ml `libpcap` is a library that allows the reading of PCAP files from wireshark. I don't know of an alternative for this on Windows. I don't really even know how the ROS2 on Windows ecosystem works so any help with this would be appreciated.

kinger-ml, 2020-12-03:
@JWhitleyWork `winpcap` is the windows alternative for `libpcap`. Currently, I have created a non-ROS application to capture the packets from Velodyne lidar and it is working correctly. I will check if using winpcap, I can resolve this issue during the ROS installation.