ros-drivers/velodyne #361: ROS Dual VLP32-C Launch Error

christopherholmes, 2020-09-02:
Hello, 

I'm trying to launch two Velodyne VLP32-C sensors in ROS using the following launch file with the command ``` roslaunch velodyne_pointcloud VLP-32C_points_multi.launch ``` which I have created.

Launching both sensors is unsuccessful and results in the below error which looks to originate from the nodelet manager. Full Error output shown further down.

```txt
[ INFO] [1599042206.074355525]: Opening UDP socket: port 2368
bind: Address already in use
[ WARN] [1599042207.078048586]: Velodyne poll() timeout
[ERROR] [1599042207.078210348]: DriverNodelet::devicePoll - Failed to poll device.
```

Both sensors are connected through a switch, with each sensor renamed as 192.168.1.201 and 192.168.1.202 respectively and I'm able to successfully communicate and launch each sensor independently using the (VLP-32C_points.launch) file. 

I've previously used this method successfully with the VLP-16 sensor. Could you advise the best way to achieve this with the VLP32-C sensor?

**Multi Sensor Launch File**
```xml
<!-- -*- mode: XML -*- -->
<!-- run velodyne_pointcloud/TransformNodelet in a nodelet manager for an VLP-32C -->

<launch>

<group sensor="sensor_1"> 
  <!-- declare arguments with default values -->
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VeloView-VLP-32C.yaml"/>
  <arg name="device_ip" default="192.168.1.201" />
  <arg name="frame_id" default="velodyne_first" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="200.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />
  <arg name="gps_time" default="false" />
  <arg name="cut_angle" default="-0.01" />
  <arg name="timestamp_first_packet" default="false" />
  <arg name="laserscan_ring" default="-1" />
  <arg name="laserscan_resolution" default="0.007" />
  <arg name="organize_cloud" default="false" />

  <!-- start nodelet manager and driver nodelets -->
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="32C"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
    <arg name="gps_time" value="$(arg gps_time)"/>
    <arg name="cut_angle" value="$(arg cut_angle)"/>
    <arg name="timestamp_first_packet" value="$(arg timestamp_first_packet)"/>
  </include>

  <!-- start transform nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/transform_nodelet.launch">
    <arg name="model" value="32C"/>
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="fixed_frame" value="" />
    <arg name="target_frame" value="" />
    <arg name="max_range" value="$(arg max_range)"/>
[... 265 lines cut ...]
[ INFO] [1599042206.031494006]: publishing 151 packets per scan
[ INFO] [1599042206.041895576]: Cut at specific angle feature deactivated.
[ INFO] [1599042206.062583081]: Reconfigure Request
[ INFO] [1599042206.065952566]: Reconfigure request.
[ INFO] [1599042206.066026178]: Target frame ID now: 
[ INFO] [1599042206.066060712]: Fixed frame ID now: 
[ INFO] [1599042206.066162489]: Initialized container with min_range: 0.4, max_range: 200, target_frame: , fixed_frame: , init_width: 0, init_height: 1, is_dense: 1, scans_per_packet: 384
[ INFO] [1599042206.066912386]: expected frequency: 9.980 (Hz)
[ INFO] [1599042206.074272106]: Only accepting packets from IP address: 192.168.1.201
[ INFO] [1599042206.074355525]: Opening UDP socket: port 2368
bind: Address already in use
[ WARN] [1599042207.078048586]: Velodyne poll() timeout
[ERROR] [1599042207.078210348]: DriverNodelet::devicePoll - Failed to poll device.
[ WARN] [1599042208.079492765]: Velodyne poll() timeout
[ERROR] [1599042208.079572132]: DriverNodelet::devicePoll - Failed to poll device.
[ WARN] [1599042209.080711556]: Velodyne poll() timeout
[ERROR] [1599042209.080768432]: DriverNodelet::devicePoll - Failed to poll device.

```

**System**
 - Ubuntu 16.04
 - ROS Version: Kinetic

Thanks.

JWhitleyWork, 2020-09-02:
@christopherholmes You can not launch two instances of the driver that are both using the same port (regardless of IP address) unless they are on different network interfaces. The driver binds to UDP port 2368 (by default) on the network port that has access to the IP address of the sensor. When you launch the 2nd copy of the driver, it also tries to bind to that port and you get the error `bind: Address already in use`.

To solve this problem, log into one of the Velodyne's web interfaces and change the Broadcast Port to something other than 2368 (the default data port) and change the Telemetry Port to something other than 2369 (the default telemetry port). Once you've power-cycled that device, change the ports for that sensor in your launch file and it should work.

christopherholmes, 2020-09-03:
@JWhitleyWork thanks for the quick reply and advice on this. I have since noticed this has been covered also in the following issue:- https://github.com/ros-drivers/velodyne/issues/108

Apologies for repetition, will close and remove.

Thanks again.

closed 2020-09-03 by christopherholmes

urbansound8K, 2022-08-27:
@christopherholmes Hi there.
wondering how can I send the output of the Velodyne node via UDP? could you please see this [https://github.com/ros-drivers/velodyne/issues/472](url)