ros-drivers/velodyne #363: Possible wrong `scan_start_time` used in function `unpack()`?

mcamurri, 2020-09-09:
- OS and Version: Ubuntu 18.04
 - ROS Version: Melodic
 - Built from Source or Downloaded from Official Repository: Official Repository
 - Version: fd07c0aed02a7

**Describe the bug**
The `unpack()` function requires as third argument the start time of the scan. I would assume this time is the timestamp of the first packet. However, when it is invoked [here](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/conversions/transform.cc#L140), the timestamp of the scan message is passed instead:
```
      data_->unpack(scanMsg->packets[i], *container_ptr, scanMsg->header.stamp);
```

When I replay a rosbag and analyze the topic `/velodyne_packets`, a message would start and end like this:
```
header: 
  seq: 5947
  stamp: 
    secs: 1593775632
    nsecs:  46826839
  frame_id: "lidar"
packets: 
  - 
    stamp: 
      secs: 1593775631
      nsecs: 947262764
data: [255, 238, 175, 34, 131, 4, 7, 0, 7, 34, 33, 5, 7, 252, 6, 31, 212, 5, 6, 2, 7, 32, 195, 6, 21, 5, 7, 43, 14, 7, 30, 6, 7, 38, 5, 7, 

<skip to the last packet...>

  - 
    stamp: 
      secs: 1593775632
      nsecs:  46826839
    data: [255, 238, 5, 34, 128, 4, 6, 5, 7, 39, 33, 5, 4, 2, 7, 45, 214, 5, 6, 0, 7, 40, 223, 6, 6, 8, 7, 43, 15, 7, 28, 16, 7, 45, 8, 7, 40, 31, 7, 32, 11, 7, 45, 35, 7, 25, 3, 7, 42, 48, 7, 33, 127, 4, 7, 255, 6, 45, 39, 5, 4, 4, 7, 42, 216, 5, 6, 255, 6, 42, 222, 6, 7, 9, 7, 45, 19, 7, 33, 15, 7, 35, 5, 7, 43, 30, 7, 28, 7, 7, 44, 37, 7, 27, 5, 7, 39, 47, 7, 35, 255, 238, 46, 34, 132, 4, 7, 251, 6, 45, 25, 5, 4, 5, 7, 39, 212, 5, 5, 0, 7, 38, 224, 6, 7, 7, 7, 45, 17, 7, 31, 16, 7, 36, 10, 7, 46, 19, 7, 27, 1, 7, 45, 34, 7, 26, 8, 7, 44, 46, 7, 28, 132, 4, 6, 0, 7, 41, 29, 5, 6, 2, 7, 42, 220, 5, 6, 3, 7, 41, 223, 6, 6, 252, 6, 46, 14, 7, 20, 19, 7, 32, 13, 7, 50, 24, 7, 30, 252, 6, 46, 31, 7, 22, 8, 7, 44, 52, 7, 30, 255, 238, 85, 34, 136, 4, 7, 1, 7, 31, 21, 5, 6, 1, 7, 44, 220, 5, 6, 4, 7, 39, 228, 6, 6, 1, 7, 42, 10, 7, 23, 10, 7, 26, 14, 7, 41, 28, 7, 28, 255,
...
```
So the header of the `velodyne_msgs::VelodyneScan` has the timestamp of the **last** packet, not the first one.

This means that [this line](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/lib/rawdata.cc#L279) will produce always a negative or zero result:
```
float time_diff_start_to_this_packet = (pkt.stamp - scan_start_time).toSec();
```

Is this the intended behavior?
Should [this line](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/conversions/transform.cc#L140) be instead:
```
data_->unpack(scanMsg->packets[i], *container_ptr, scanMsg->packets[0].stamp);
```

As a final remark, I would suggest to remove dependencies on ROS data structures from the `unpack()` function. It's computation is completely ROS-agnostic, so its signature could easily be:
```
void unpack(const raw_packet_t* raw,  const uint64_t& pkt_time,  ContainerBase& data, const uint64_t& scan_start_time);
```
without zero modifications to the logic. This way the `unpack()` could be used by non-ROS programs.

JWhitleyWork, 2020-09-11:
@mcamurri This actually depends on your configuration of the sensor. See the lines in the driver [starting here](https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/driver/driver.cc#L259) where tthe timestamp of the scan is chosen to either be the timestamp of the first packet or the last. This is based on the parameter `timestamp_first_packet` which can be configured in all of the launch files.

Please feel free to re-open the ticket if you think there is still an issue.

closed 2020-09-11 by JWhitleyWork

mcamurri, 2020-09-12:
Thank you @JWhitleyWork for the explanation about the `timestamp_first_packet` driver option.

Can you please confirm that the `unpack()` function works correctly only when it invoked with the start of the scan timestamp (i.e., when it is fed by a driver with the `timestamp_first_packet` set to `true`)? 

In that case, calling `unpack()` with `scanMsg->packets.front().stamp` as argument would be a better option than using a timestamp you don't know whether it is the start or the end of the scan. 

There is no conditional invocation of `unpack()` inside `transform.cc` based on `timestamp_first_packet`, so the function is always invoked assuming that the timestamp of the scan message is also the start of the scan.