ros-drivers/velodyne #368: New buildfarm release for Noetic

Timple, 2020-11-09:
I'm trying to release the `points_preprocessor` package which depends on `velodyne_pcl`. However compilation fails due to a typo fixed in #357 

Could a new release be triggered which includes this fix?

JWhitleyWork, 2020-11-10:
Done: https://github.com/ros/rosdistro/pull/27285

closed 2020-11-10 by JWhitleyWork

pedroexenberger, 2020-11-27:
Hi!
I'm also facing the problem fixed in #357. I just `apt update` and `apt install` the `ros-noetic-velodyne-pcl`, but it says 
`ros-noetic-velodyne-pcl is already the newest version (1.6.0-1focal.20200930.012136).` But I'd like to have version 1.6.1-1.

I'm not familiar with the ros distribution workflow, so what ros/rosdistro#27285 means exactly? I saw it was 'merged' with ros master, but I still cannot install the 1.6.1-1 version through apt. 

Do you have any tips for me to update the package to the fixed version? Thanks!