ros-drivers/velodyne #369: How can I build "velodyne-driver" at ros2(Dashing)

JaredLee1218, 2020-11-10:
I am the newby in ros2. 

How can I use my VLP-16 velodyne lidar with ros2 dashing? 

I do not know how do I start first...There isn't any hint to newby...

JWhitleyWork, 2020-11-11:
The driver for `ros2 dashing` has been released to the build farm which means that binary versions are available so you don't have to build it from source! If you're using Ubuntu, just run `sudo apt install ros-dashing-velodyne` and you'll get it fully installed to `/opt/ros/dashing`. After that, run `source /opt/ros/dashing/setup.bash`. Once that's done, you can use the launch files in the `velodyne` package with `ros2 launch`. If your VLP-16 is using the default IP address of 192.168.1.201 and your network interface is set up to be on the same subnet, that's all you'll have to do. Otherwise, you'll need to create your own launch file which looks like the examples [here](https://github.com/ros-drivers/velodyne/tree/dashing-devel/velodyne/launch) and modify the parameters for the `velodyne_driver` node to use the IP address of your sensor.

If you have further questions, please use [ROS Answers](https://answers.ros.org/) so the whole community can respond and see the solution.

closed 2020-11-11 by JWhitleyWork

SyahirMuzni, 2022-09-08:
Hi there! If I build it from binary, how do I edit the launch file if I want to configure some settings for the lidar?