ros-drivers/velodyne #372: Version 1.6.1-1 still not available?

pedroexenberger, 2020-11-27:
Hi!
I have a problem related to the issue https://github.com/ros-drivers/velodyne/issues/368 (closed), but I thought it would be better opening a new one.

I'm also facing the problem fixed in #357, which was fixed on version 1.6.1-1. However, I just `apt update` and `apt install` the `ros-noetic-velodyne-pcl`, but it says `ros-noetic-velodyne-pcl is already the newest version (1.6.0-1focal.20200930.012136)`. So we are still in version 1.6.0-1?

@JWhitleyWork pointed this -> ros/rosdistro#27285 as a solution.

I'm not familiar with the ros package distribution/release workflow, so what ros/rosdistro#27285 means exactly? I saw it was 'merged' with ros master, but I still cannot install the 1.6.1-1 version through apt. Is the updated package still not available?

Do you have any tips for me to update the package to the fixed version? 

Thanks!

Timple, 2020-11-27:
According to this you can expect a release on the 30th:

https://discourse.ros.org/t/preparing-for-noetic-sync-2020-11-30/17574

pedroexenberger, 2020-11-27:
Great! Thanks @Timple for your valuable feedback.

closed 2020-11-27 by pedroexenberger