ros-drivers/velodyne #373: No libdata_containers.so when installing noetic package?

pedroexenberger, 2020-12-01:
**Quick description**

When installing the `ros-noetic-velodyne` through apt/apt-get there is no libdata_containers.so generated on the default lib folder for ros-noetic (`/opt/ros/noetic/lib`).

**Detailed description**

A recent pull request (https://github.com/ros-drivers/velodyne/pull/359) was merged to this repo allowing external use of the class implementations for DataContainerBase (currently `velodyne_pointcloud::PointcloudXYZIRT` and `velodyne_pointcloud::OrganizedCloudXYZIRT`).

This new feature was reported to be part of the package version 1.6.1-1 for ROS noetic https://github.com/ros/rosdistro/pull/27285, which was [sync'd yesterday](https://discourse.ros.org/t/preparing-for-noetic-sync-2020-11-30/17574).

However, if one install the new `ros-noetic-velodyne` package through apt/apt-get there is no `libdata_containers.so` generated on the default lib folder for ros-noetic `/opt/ros/noetic/lib`, although other .so files related to the velodyne package are there (e.g., `libvelodyne_rawdata.so`).

Alternatively, if one clone this repo, run `catkin_make` and `catkin_make install`, then the `libdata_containers.so` _is_ generated in the `install/lib` folder of the catkin workspace, and one can use the previously mentioned DataContainer types. Shouldn't this also happen when installing with the noetic-package?

Also, running pkg-config on the repo-based installation `pkg-config ~/catkin_ws/install/lib/pkgconfig/velodyne_pointcloud.pc --libs` do show the `-ldata_containers`linked, whereas running it against the apt installed package `pkg-config velodyne_pointcloud --libs` does not show the link with libdata_containers.so.

Is this a bug? Or is this expected to be like this? I'm not familiar with the release of ROS packages (I'm mostly a user) so I'm not sure if this is the desired behavior, but it would be great if we could use the 'standard' apt package to have this new functionality, instead of the repo clone/build approach.

OS: Ubuntu 20.04.1
Package version reported on apt-get install: ros-noetic-velodyne_1.6.1-1focal.20201110.162946_amd64.deb

JWhitleyWork, 2020-12-01:
Running the following test in a Docker container results in the file being installed correctly:
```bash
$ docker pull ros:noetic
$ docker run --rm -it ros:noetic
noetic$ apt-get update && apt-get install -y ros-noetic-velodyne
noetic$ ls -la /opt/ros/noetic/lib/*data_containers*
```

Result:
```bash
-rw-r--r-- 1 root root 113728 Nov 10 16:22 /opt/ros/noetic/lib/libdata_containers.so
```

Version:
```bash
ros-noetic-velodyne/focal,now 1.6.1-1focal.20201110.162946 amd64 [installed]
```

pedroexenberger, 2020-12-01:
Thanks, @JWhitleyWork.

For some reason, the package was not upgrading correctly with the apt-get install only. It was solved with `apt-get purge ros-noetic-velodyne*` and then re-installed everything.

closed 2020-12-01 by pedroexenberger

JWhitleyWork, 2020-12-01:
Glad it was resolved!