ros-drivers/velodyne #374: Timeline for gps_time feature on ros-melodic-velodyne package

syigzaw, 2020-12-05:
I'm currently using ros-melodic-velodyne version 1.5.2. I've looked at [https://github.com/ros-drivers/velodyne/issues/341](url), [https://github.com/ros-drivers/velodyne/issues/340](url), [https://github.com/ros-drivers/velodyne/pull/347](url), which have led me to understand that this is the last release so far for the ros-melodic-velodyne package. I'm trying to use one of the earliest feature additions after 1.5.2, the gps_time parameter. Is there a way to add this to my local version of ros-melodic-velodyne or do I have to wait for the official ROS package to be updated? And if waiting is what's necessary, is there a time frame that you think this could be up by? Thanks in advance for the help.

JWhitleyWork, 2020-12-05:
@syigzaw A `melodic` release is currently waiting on @spuetz to finish #347.

JWhitleyWork, 2020-12-05:
As far as using the latest features, you can use the `master` branch and build it locally - you may just have problems if any of your software depends on `velodyne_pointcloud/include/velodyne_pointcloud/point_types.h` to exist and not have changed.

syigzaw, 2020-12-05:
> As far as using the latest features, you can use the `master` branch and build it locally - you may just have problems if any of your software depends on `velodyne_pointcloud/include/velodyne_pointcloud/point_types.h` to exist and not have changed.

Thanks for the quick response. I don't see any instructions in the readme on how to build locally. How would I go about doing that? Thanks again.

JWhitleyWork, 2020-12-07:
@syigzaw If you have not yet created a ROS workspace, you will need to do so. See http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment#Create_a_ROS_Workspace for details. Once you have created your workspace, clone the `velodyne` repository into the `src` folder and build the workspace with `catkin_make`. Once that is done, source the workspace with a command similar to `source ~/your_workspace/devel/setup.bash` and you should then be able to run the version of the driver in your workspace.