ros-drivers/velodyne #375: VLP32MR support

sasupe, 2020-12-09:
Hello,
Can VLP32MR be used with this driver code?
I see VLP32MR is similar to VLP32C except for the range.

Can the param file at https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/params/VeloView-VLP-32C.yaml be used for VLP32MR too?
Are there any other things to be considered.

Thank you for your support.

JWhitleyWork, 2020-12-11:
I'm actually not sure. However, Velodyne should have provided an XML configuration file for the lidar for VeloView. If they did not, please request it from them. Once you have that configuration file, you can use the [gen_calibration.py script](https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/scripts/gen_calibration.py) to generate a calibration that is compatible with this driver from it. We would then really appreciate a Pull Request to add that file to the driver.

Beyond the calibration file, I think it is unlikely that they changed how the packets are sent which means there shouldn't be any changes required to the code (I hope). I recommend testing with the new YAML file and setting the driver to the VLP-32C model to see if the data look correct. If not, additional changes may be required. Let me know the results of your testing and I'll be glad to make any configuration changes to enable easy use of the sensor.

sasupe, 2020-12-13:
Thank you for your quick reply. 
I contacted Velodyne support and they have confirmed that VLP-32c xml file can be used with VLP-32MR since they have the same azimuth offsets and vertical angles.  

Closing the issue.

closed 2020-12-13 by sasupe