ros-drivers/velodyne #377: YAML-CPP exception when loading VLP16db.yaml

ds-ssj, 2020-12-17:
** Environment ** 
- OS and Version: Ubuntu 18.04
 - ROS Version: ROS2 dashing
 - Built from Source or Downloaded from Official Repository:
 - Version: 2.0.0-1bionic.20201125.071158

**Describe the bug**
When run `velodyne_transform_node` by using `VLP16db.yaml` provided under the repo, an exception shows that:
```
[INFO] [velodyne_transform_node]: correction angles: ./VLP16db.yaml
terminate called after throwing an instance of 'std::runtime_error'
  what():  YAML Exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
```

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo apt install ros-dashing-velodyne`
2. source ros2 the setup script
3. write a ROS2 `velodyne_transform_param.yaml` config file like that:
```
---
velodyne_transform_node:
  ros__parameters:
    min_range: 0.9
    calibration: "./VLP16db.yaml"
```
4. run by the command `ros2 run velodyne_pointcloud velodyne_transform_node __params:=./velodyne_transform_param.yaml`

**Expected behavior**
Run it correctly

**Additional context**
Add any other context about the problem here.

ds-ssj, 2020-12-17:
> ** Environment **
> 
> * OS and Version: Ubuntu 18.04
> * ROS Version: ROS2 dashing
> * Built from Source or Downloaded from Official Repository:
> * Version: 2.0.0-1bionic.20201125.071158
> 
> **Describe the bug**
> When run `velodyne_transform_node` by using `VLP16db.yaml` provided under the repo, an exception shows that:
> 
> ```
> [INFO] [velodyne_transform_node]: correction angles: ./VLP16db.yaml
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  YAML Exception: yaml-cpp: error at line 0, column 0: invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
> ```
> 
> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. `sudo apt install ros-dashing-velodyne`
> 2. source ros2 the setup script
> 3. write a ROS2 `velodyne_transform_param.yaml` config file like that:
> 
> ```
> ---
> velodyne_transform_node:
>   ros__parameters:
>     min_range: 0.9
>     calibration: "./VLP16db.yaml"
> ```
> 
> 1. run by the command `ros2 run velodyne_pointcloud velodyne_transform_node __params:=./velodyne_transform_param.yaml`
> 
> **Expected behavior**
> Run it correctly
> 
> **Additional context**
> Add any other context about the problem here.

I tests the yaml-cpp lib v6.0, 5.3, 5.2, 5.0

nshahrestani, 2020-12-18:
Try providing the full yaml file path and not appending a period at the beginning.