ros-drivers/velodyne #379: [WARN]  :  No Azimuth Cache configured for model VLP16

he-guo, 2020-12-21:
hello, I have this problem, and how do I configure it

JWhitleyWork, 2020-12-22:
@he-guo Needs more context. Please provide fields from the Bug Report template:

**Please complete the following information:**
 - OS and Version: [e.g. Ubuntu 16.04]
 - ROS Version: [e.g. Kinetic]
 - Built from Source or Downloaded from Official Repository:
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

hunter12580, 2021-02-23:
are you using jetson tx2? I've met the same problem on tx2, but not on windows pc (with vlp-16) @he-guo

GoodJobBS, 2021-06-11:
Have you solved this problem? I also encountered this problem in the ROS system