ros-drivers/velodyne #385: [velodyne_pointcloud]

YoshuaNava, 2021-01-21:
**Please complete the following information:**
 - OS and Version: Ubuntu 20.04
 - ROS Version: Noetic
 - Built from Source
 - Version: source, latest master.

**Describe the bug**
By reading through the VLP-16 manual I found the following table that specifies vertical corrections, to be applied to all sensors from this family:

[image]

I quickly skimmed through the code of velodyne_pointcloud and found that we do apply a vertical correction when transforming range measurements into points. Those corrections are loaded from the calibration files under the folder params/.

When I checked the values of the VLP16 vertical corrections, they were completely different to what is specified in the manual:

https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/params/VLP16db.yaml#L4

Is this expected?

Could it be that the vertical corrections represented in the calibratio file have undergone an additional math operation? e.g. change of unit, rotation, etc.

YoshuaNava, 2021-02-01:
@JWhitleyWork Sorry for bothering. Do you know if these corrections are actively applied to the VLP-16 data?

kamcnally, 2021-02-04:
Are you referring to the `vert_correction` field within the yaml file?

If so, according to [REP 103](https://www.ros.org/reps/rep-0103.html), the angle you are seeing in this file is in radians whereas the manual is in degrees.

If I take the values from the manual for the column `Vertical Angle VLP-16` and convert to radians  you get the following which all match up so look correct to me.

Driver | Manual (deg) | Manual (rad)
-- | -- | --
-0.2617993878 | -15 | -0.2617993878
0.01745329252 | 1 | 0.01745329252
-0.2268928028 | -13 | -0.2268928028
0.05235987756 | 3 | 0.05235987756
-0.1919862177 | -11 | -0.1919862177
0.0872664626 | 5 | 0.0872664626
-0.1570796327 | -9 | -0.1570796327
0.1221730476 | 7 | 0.1221730476
-0.1221730476 | -7 | -0.1221730476
0.1570796327 | 9 | 0.1570796327
-0.0872664626 | -5 | -0.0872664626
0.1919862177 | 11 | 0.1919862177
-0.05235987756 | -3 | -0.05235987756
0.2268928028 | 13 | 0.2268928028
-0.01745329252 | -1 | -0.01745329252
0.2617993878 | 15 | 0.2617993878

YoshuaNava, 2021-02-04:
Hi @kamcnally,
Thank you for your answer.

The values you mention specify the angle of the beams with respect to the horizon, right? That's correct in the calibration file, I agree :+1: 

However, in my issue post I mention that in the manual some vertical corrections (in mm) are specified in the VLP-16 manual. According to the manual, these corrections compensate the difference in the origin of the sensor frame and the laser and should be removed from the X,Y,Z points in the sensor frame:

[image]

tvalimaki, 2021-02-04:
I think this is a duplicate of #190 referring to the `vert_offset_correction`.

YoshuaNava, 2021-02-21:
@tvalimaki Thank you for your reply. You are right. I'll close this issue as duplicate.

closed 2021-02-21 by YoshuaNava