ros-drivers/velodyne #387: No velodyne_pointcloud/point_types.h ?

plumewind, 2021-01-22:
**Please complete the following information:**
 - OS and Version: [Ubuntu 12004]
 - ROS Version: [noetic]
 - Built from Source or Downloaded from Official Repository:[Official Repository]
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]
```
sudo dpkg -s ros-noetic-velodyne 
Package: ros-noetic-velodyne
Status: install ok installed
Priority: optional
Section: misc
Installed-Size: 13
Maintainer: Josh Whitley <josh.whitley@autoware.org>
Architecture: amd64
Version: 1.6.1-1focal.20201207.154637
Depends: ros-noetic-velodyne-driver, ros-noetic-velodyne-laserscan, ros-noetic-velodyne-msgs, ros-noetic-velodyne-pointcloud
Description: Basic ROS support for the Velodyne 3D LIDARs.
Homepage: http://www.ros.org/wiki/velodyne
```
**Describe the bug**
get a error: fatal error: velodyne_pointcloud/point_types.h

**To Reproduce**
Steps to reproduce the behavior:
go the path `velodyne_pointcloud/point_types.h`， there is no file.

Anyone hel me , thanks .

JWhitleyWork, 2021-01-28:
The file was moved from `velodyne_pointcloud` to `velodyne_pcl` on the `noetic` branch some time ago. I have confirmed that 1.6.1 has the file in `velodyne_pcl` and nowhere in the code is it referenced in `velodyne_pointcloud`. Please uninstall `ros-noetic-velodyne` and all of its dependencies and make sure you don't have a copy of the `velodyne` repository somewhere in a workspace that may be overriding the installed one in `/opt/ros/noetic`.

charvi-077, 2022-06-30:
Thanks @JWhitleyWork it resolved our issue also, in noetic the pkg is velodyne_pcd only instead of velodyne_pointcloud which consists the point_types.h file and also there the message type defined is PointXYZIRT, so we may need to change that as well because in some packages  PointXYZIR is used.