ros-drivers/velodyne #389: Velodyne velarray Lidar support

Magmanat, 2021-01-27:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I am trying to look for a solid state lidar with a wide horizontal view but yet a decently wide vertical view that has a ros driver for obstacle avoidance for my robot
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Is anyone developing a ros driver for these new velarray sensors from velodyne velabit to m1600? 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Another lidar which i would use other than this might be the leddar pixel but i am not too sure if that has a ros driver either
**Additional context**
Add any other context or screenshots about the feature request here.

JWhitleyWork, 2021-01-27:
The driver currently only supports spinning lidars. We would need someone with the hardware to develop the support for additional sensors.

JKTesla, 2021-02-23:
> The driver currently only supports spinning lidars. We would need someone with the hardware to develop the support for additional sensors.

Hi buddy, do you have independently velarray Lidar ros driver(base on Ubuntu16.04 and kinetic) to share me? We are testing velarray, hope to record data using rosbag.

jacknpainter, 2022-01-10:
Has their been any update with this issue? Are there any ROS drivers at all for the Velarray LIDARs?

Chick92, 2022-03-29:
HAs their been any update? I have a Velarray M1600 sat on my desk if a test article is required.

JWhitleyWork, 2022-03-29:
No pull requests have been made to support this functionality. Since this is a community-developed driver, someone with both access to the hardware and software development experience would need to add this feature and make a pull request. I am happy to review any such pull request but I don't have access to the hardware nor the time to add the feature myself.

fjahncke, 2023-01-12:
For people looking for a M1600 driver, there are drivers available directly from Velodyne in the Vella SDK https://velodynelidar.com/products/vella/.

ArseniiPavlov, 2023-06-09:
> For people looking for a M1600 driver, there are drivers available directly from Velodyne in the Vella SDK https://velodynelidar.com/products/vella/.

This page is not available, is there any other option to get this software?

fjahncke, 2023-06-09:
> > For people looking for a M1600 driver, there are drivers available directly from Velodyne in the Vella SDK https://velodynelidar.com/products/vella/.
> 
> This page is not available, is there any other option to get this software?

Yes they changed the website you can now register here: https://www.vellaportal.com/register

ArseniiPavlov, 2023-06-09:
> Yes they changed the website you can now register here: https://www.vellaportal.com/register

@fjahncke Thank you very much!

mbiselx, 2023-08-16:
I wrote a small ROS driver just for the M1600 : https://github.com/mbiselx/velarray
WIth a bit of tweaking, it could also work for the 800s (I have no idea about the velabits though). 
Unfortunately my skillz are not good enough to attempt integrating this into this project (nor do I have the time right now 😢 ). If anyone wants to try porting it, they are quite welcome.