ros-drivers/velodyne #392: Per-point timestamps sometimes out of order

dhellfeld, 2021-03-03:
- Ubuntu 20 (Docker)
 - ROS Noetic
 - Built using `apt-get install ros-noetic-velodyne` in a docker built on top of the official ROS docker: `noetic-ros-base`

I recorded a bag with a VLP16 and noticed there are a few scans in which the per-point timestamps are out of order (meaning there is a point in the cloud that has a timestamp larger than the last point in the cloud). This causes the SLAM algorithm I am using (Google Cartographer) to crash.

I'm not sure this is related, but I also changed the "rpm" roslaunch arg to change how many UDP packets are assembled into one cloud (changed from 600 to 4800 to effectively return 8 pc2 messages in one sweep). This is the only parameter I changed in the launch file. Also, why is this roslaunch arg called "rpm"? It does not change the rotation rate, like the "rpm" parameter in the Velodyne configuration.

Any thoughts here? Or is there a way to turn the per-point timestamp off?

hengjiUSTC, 2021-06-11:
Same issue, haven't find solution yet

jaeseok4104, 2023-06-06:
how can you get timestamp of per-point?