ros-drivers/velodyne #395: Velodyne output does not contain full pointcloud

osholsonel20, 2021-03-08:
Device: HDL64_ES2
OS: Ubuntu 20.04

Device RPM: 900

I'm trying to receive a message from the device that I can use the data to overlay onto our fisheye cameras. However, when trying to export out a .pcd file of one single ros message, I notice that the pointcloud is cut in half. Should it be publishing the full pointcloud every ros message? I'm a bit confused as to why I can't save the entire cloud.

[image]

osholsonel20, 2021-03-08:
I also notice I get errors in the diagnostic message saying that my Hz is too fast. The Hz i'm getting is around 22 - 24, and the expected hz is from 12 - 15. Not sure if that is an issue here as well.

JWhitleyWork, 2021-03-09:
@osholsonel20 Have you modified the launch file to reflect your rotation rate? The default is 600 and is set as a parameter in the launch file.

osholsonel20, 2021-03-09:
I have, changed from default 600 to 900. When it was at 600 with a 900 RPM sensor, a small chunk, about 5 degrees was missing from the point cloud message. 

I've also used VeloView to test the device and it stated it was running at 900 RPM, just to confirm.

[image]

osholsonel20, 2021-03-09:
Here is the launch file we are using.

[64e_points.txt](https://github.com/ros-drivers/velodyne/files/6109047/64e_points.txt)

osholsonel20, 2021-03-09:
Source
Branch: Master
Commit Hash: e20b5e3
July 17th, 2020

osholsonel20, 2021-03-22:
@JWhitleyWork Any updates on your end? Looking to try and fix our Velodyne device to get full point cloud fidelity for our data labeling team.

osholsonel20, 2021-04-21:
@JWhitleyWork I've brought the device down to 600 RPM. Do you know how many ros messages makes up one entire pointcloud? That's really what we need is just to save out a complete cloud as a .pcd file.