ros-drivers/velodyne #399: failed to load nodelet

wang19950831, 2021-04-14:
When I run roslaunch velodyne_pointcloud VLP16_points.launch, the following error appears. Does anyone know the reason?

started roslaunch server http://localhost:45527/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/gps_time: False
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_driver/timestamp_first_packet: False
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1
 * /velodyne_nodelet_manager_transform/calibration: /home/ldx/catkin_...
 * /velodyne_nodelet_manager_transform/fixed_frame: 
 * /velodyne_nodelet_manager_transform/max_range: 130.0
 * /velodyne_nodelet_manager_transform/min_range: 0.4
 * /velodyne_nodelet_manager_transform/model: VLP16
 * /velodyne_nodelet_manager_transform/organize_cloud: False
 * /velodyne_nodelet_manager_transform/target_frame: 

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)
    velodyne_nodelet_manager_transform (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [20410]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1690751c-9cc5-11eb-ac61-48b02d05b5eb
process[rosout-1]: started with pid [20422]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [20429]
process[velodyne_nodelet_manager_driver-3]: started with pid [20430]
process[velodyne_nodelet_manager_transform-4]: started with pid [20431]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [20436]
[ INFO] [1618365588.098313350]: Initializing nodelet with 8 worker threads.
[ INFO] [1618365588.128788168]: correction angles: /home/ldx/catkin_velodyne/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml
[ INFO] [1618365588.136973169]: laser_ring[ 0] =  0, angle = -0.261799
[ INFO] [1618365588.137077590]: laser_ring[ 2] =  1, angle = -0.226893
[ INFO] [1618365588.137123193]: laser_ring[ 4] =  2, angle = -0.191986
[ INFO] [1618365588.137170588]: laser_ring[ 6] =  3, angle = -0.157080
[ INFO] [1618365588.137228383]: laser_ring[ 8] =  4, angle = -0.122173
[ INFO] [1618365588.137277057]: laser_ring[10] =  5, angle = -0.087266
[ INFO] [1618365588.137317028]: laser_ring[12] =  6, angle = -0.052360
[ INFO] [1618365588.137362246]: laser_ring[14] =  7, angle = -0.017453
[ INFO] [1618365588.137401256]: laser_ring[ 1] =  8, angle = +0.017453
[ INFO] [1618365588.137464108]: laser_ring[ 3] =  9, angle = +0.052360
[ INFO] [1618365588.137533552]: laser_ring[ 5] = 10, angle = +0.087266
[ INFO] [1618365588.137646166]: laser_ring[ 7] = 11, angle = +0.122173
[ INFO] [1618365588.137734107]: laser_ring[ 9] = 12, angle = +0.157080
[ INFO] [1618365588.137927846]: laser_ring[11] = 13, angle = +0.191986
[ INFO] [1618365588.138022699]: laser_ring[13] = 14, angle = +0.226893
[ INFO] [1618365588.138068430]: laser_ring[15] = 15, angle = +0.261799
[ INFO] [1618365588.138521191]: Number of lasers: 16.
[ INFO] [1618365588.162725580]: Reconfigure request.
[ INFO] [1618365588.162924599]: Target frame ID now: 
[ INFO] [1618365588.163006875]: Fixed frame ID now: 
[ INFO] [1618365588.163146787]: Initialized container with min_range: 0.4, max_range: 130, target_frame: , fixed_frame: , init_width: 0, init_height: 1, is_dense: 1, scans_per_packet: 384
free(): invalid pointer
[FATAL] [1618365588.542439349]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1618365588.543011829]: Failed to load nodelet '/velodyne_nodelet_manager_transform` of type `velodyne_pointcloud/TransformNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1618365588.543348104]: Failed to load nodelet '/velodyne_nodelet_manager_laserscan` of type `velodyne_laserscan/LaserScanNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 20429, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager-2.log].
log file: /home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager-2*.log
[velodyne_nodelet_manager_driver-3] process has died [pid 20430, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_driver-3.log].
log file: /home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_driver-3*.log
[velodyne_nodelet_manager_transform-4] process has died [pid 20431, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load velodyne_pointcloud/TransformNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_transform __log:=/home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_transform-4.log].
log file: /home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_transform-4*.log
[velodyne_nodelet_manager_laserscan-5] process has died [pid 20436, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load velodyne_laserscan/LaserScanNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_laserscan __log:=/home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_laserscan-5.log].
log file: /home/ldx/.ros/log/1690751c-9cc5-11eb-ac61-48b02d05b5eb/velodyne_nodelet_manager_laserscan-5*.log

JWhitleyWork, 2021-04-14:
The issue causing the nodelet to crash is: `free(): invalid pointer`. However, in order to help troubleshoot this, we need more context. Please provide:

- Operating system and version
- ROS version
- Version/commit that you are using
- Building from source or installing from binaries?

wang19950831, 2021-04-15:
> The issue causing the nodelet to crash is: `free(): invalid pointer`. However, in order to help troubleshoot this, we need more context. Please provide:
> 
> * Operating system and version
> * ROS version
> * Version/commit that you are using
> * Building from source or installing from binaries?

Thank you for the help.
I'm using Ubuntu18.04
ros melodic
Do you mean the firmware version? it is 3.0.41.1.
Building from source code here. But maybe i operate 'sudo apt-get install ros-melodic-velodyne-driver'.I don't know if there is a conflict.

dresio, 2021-06-02:
I had the same error running on a jetson xavier headless over ssh. We found the issue to be something with the setup.bash file and dependence.  Our fix was in any terminal before starting the nodelet to resource the catkin_ws/devel/setup.bash file. 
The setup.bash file is sourced in the .bashrc so this theoretically shouldn't change anything, however it does fix this problem. 
Let me know if this helps!