ros-drivers/velodyne #412: Catkin_make Error

ManChrys, 2021-06-23:
- OS and Version: 18.04
 - ROS Version: melodic

Hello guys,

I tried to install velodyne package following these steps

https://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20HDL-32E

but i have the following error

[image]

Can anyo1 help me to solve this ? 

Thanks in advance!

JWhitleyWork, 2021-06-23:
This problem is an issue with other packages in your workspace and is not related to the Velodyne driver. Please ask this question on [ROS Answers](https://answers.ros.org) and you should get an answer quickly.

closed 2021-06-23 by JWhitleyWork

ManChrys, 2021-06-23:
ok thank you!