ros-drivers/velodyne #416: Velodyne poll() timeout

hxj0316, 2021-07-28:
Hello, I compiled vlp_driver under ros2, and then I ran the "ros2 run velodyne_driver velodyne_driver_node" command, and the result was as shown in the figure below. I pinged the address of the lidar, and it was able to ping. , I don’t know why there is such a problem, please help me to take a look, thank you.
[image]

TZECHIN6, 2023-04-13:
I am facing the same issue now. LiDAR can be ping and the IP setting should be correct, but still getting poll timeout.

Anyone could help on this? Thanks

himhan34, 2023-04-13:
I think i saw some reports that dealing ros2 with that file can you search closed repo?

TZECHIN6, 2023-04-13:
@himhan34 you are right, I can connect the lidar at that time... so I really have no clues why now I only setting up for one lidar and it doesn't work... (the project is new, so I haven't changed any of the code from ros2 branch)

himhan34, 2023-04-13:
mostly when i used the vlp 16 with ros1 and when i got that message i started from the bottom, 
so i started with the beginning, i installed the wireshark and check the ip of the lidar, and check to see the lidar is properly
connected and sending and getting the right packets try to start at the beginning

TZECHIN6, 2023-04-13:
thats why I am so frustrated, as I done all of these: IP config, ping test, reset all the hardware & software, wireshark to see package deliver status, as well as using veloview to see the lidar scan. But none of these indicated wheres the problem, except veloview, as it also doesn't show any lidar point as well, but other than that, everything looks reasonable to me. lidar can be pinged, wireshark can see the package sending from correct IP to correct destination IP with UDP package length 1268/1238 I forgot the number, but its corret.

TZECHIN6, 2023-04-14:
@himhan34  I fix the issue somehow

FYI, https://github.com/ros-drivers/velodyne/issues/505

himhan34, 2023-04-14:
Its good to hear that . 
Have a nice day 
Best regards

closed 2023-04-14 by JWhitleyWork