ros-drivers/velodyne #421: ros-galactic-velodyne support

dankenbrand, 2021-08-20:
I would like to switch from ros foxy to ros galactic and continue using the velodyne package.

As it did not work installing it with apt, I tried to install it natively. This resulted in more and more missing packages to install (e.g. ros-galactic-angles, ros-galactic-diagnostic-updater, ros-galactic-perception-pcl). After installing the mentioned packages, I ended up with "pkg_check_modules" missing.

Does anyone have a solution to the describe problem or are there any plans to release a galactic version soon?

JWhitleyWork, 2021-08-23:
Galactic release incoming: https://github.com/ros/rosdistro/pull/30551

For the time being, you should add the package to  your ROS workspace and build it from source. You should use `rosdep` to install any missing dependencies.

closed 2021-08-23 by JWhitleyWork

dankenbrand, 2021-09-09:
Thank you for your reply.
Concerning the Galactic release and the link you posted (https://github.com/ros/rosdistro/pull/30551) I'm confused if the sync with the velodyne package is completed. If so, I'm facing another issue using ros-galactic-velodyne ("E: Unable to locate package ros-galactic-velodyne") or if the package was not included in the sync (https://discourse.ros.org/t/new-packages-for-galactic-geochelone-2021-08-25/22032) as I can't find the package listed.
Maybe you know more?

JWhitleyWork, 2021-09-09:
The package did not make that sync and will be added in the next sync. Sorry for the confusion.

aghari19, 2021-11-16:
Hi @dankenbrand, 

I have a question with install velodyne drivers in foxy. I use this command: **sudo apt-get install ros-foxy-velodyne**
and it fails to fetch the packages. From your post my understanding is you were using velodyne in foxy. Can you please help me with the command to execute to install the velodyne drivers in foxy. 

Thank you,

dankenbrand, 2021-11-17:
Hi @aghari19,

As I remember, I did it as you mentioned with `sudo apt-get install ros-foxy-velodyne`.
Could you maybe more specific about the error you receive?