ros-drivers/velodyne #422: Incorrect timestamps (!=GPRMC string) while using gps_time = true and PPS

gabriel-earle, 2021-08-23:
**Please complete the following information:**
 - OS and Version: Ubuntu 18.04
 - ROS Version: Melodic
 - Built from Source or Downloaded from Official Repository: Built from source
 - Version: melodic-devel commit b809629

**Describe the bug**
Hello, I am getting unexpected timestamps using gps_time = true with GPRMC strings.  I am simulating GPRMC strings each second from a Teensy (Arduino) GPIO, and another pin sends a PPS.  I am monitoring the GPRMC strings and they have the right time.  However, the ROS messages have an timestamp that does not match. It's usually around 20-30 minutes early, but the offset is not constant.

I am finding the error with both a Velodyne VLP-16 and a Velodyne VLP-16 Hi-res.  Please let me know any recommendations I can try for debugging, many thanks.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone melodic-devel on Ubuntu 18.04
2. Send GPRMC strings and PPS lock to Velodyne VLP-16.
3. Run driver with gps_time = true and launch file pictured below.
4. Inspect the header timestamps.

**Expected behavior**
The velodyne_packets and velodyne_points have header timestamps matching the GPRMC string time/date fields.

**Additional context**
The issue:
[image]
So as I print the GPRMC messages coming from the microcontroller to console, the logging timestamps match (1629754973 = 9:42:53PM GMT Aug 23, 2021).

However, the timestamps from the velodyne_points headers came out ~20 minutes ahead in this case.

My launch file:
[image]

My lidar config with PPS lock confirmed:
[image]

My lidar firmware:
[image]

adthoms, 2023-06-29:
@gabriel-earle and others, this was an issue with NMEA scheduling after the PPS signal that was not documented in the VLP16 manual we were using. We have long corrected this problem, and our original issue does not concern the velodyne driver. This issue may be closed.

JWhitleyWork, 2023-06-29:
Thank you, @adthoms!

closed 2023-06-29 by JWhitleyWork