ros-drivers/velodyne #431: ROS2 Foxy communicating with Velodyne VLP16

aghari19, 2021-11-19:
Hello, 

I am using Ubuntu 20.04, running ROS2 Foxy and trying to communicate with the Velodyne VLP16. I want to know the the packages I have to run to to see the /scan, /velodyne_points topics on Rviz. Eventually I want to run slam_gmapping on it. My LiDAR is on 192.168.3.200 ip and port 2368

I git cloned [this](https://github.com/ros-drivers/velodyne.git) branch foxy-devel repo into a workspace, sourced it. 

When I run the velodyne_driver package I see this:
[image]

When I do rostopic list I see this: 
[image]

When I do echo /velodyne_packets I see the packets however I cannot see them on rviz. 

I tried running the velodyne_pointcloud package I see this:
[image]

When I do rostopic list I see /velodyne_points and I do echo but I do not see anything:
[image]

I tried running the velodyne_laserscan I see this:
[image]

When I do rostopic list I see /velodyne_points and /scan and I try to echo both the topic but I cannot see any data. 
[image]

I am not sure how to proceed. I tried the same with with the ros2 branch and I see the same things. Can someone help with letting me know the packages I need to run to be able to see topics /velodyne_points and /scan on Rviz

Thank you for the help 
Haritha

bclaw524, 2022-07-19:
I am also having a similar issue. 

I can get the topic data using ros1, but not really sure how to proceed with ros2.

bclaw524, 2022-07-19:
and in regular fashion, the moment I ask, I find the solution.

For any other newbies stuck, you'll want to use:

`ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py`

NadimDeeb, 2023-05-30:
> > I am also having a similar issue.
> > I can get the topic data using ros1, but not really sure how to proceed with ros2.
> > Any follow up on how to solve this?
> 
> and in regular fashion, the moment I ask, I find the solution.
> 
> For any other newbies stuck, you'll want to use:
> 
> `ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py`

This really helped me thank you!

closed 2023-06-01 by JWhitleyWork