ros-drivers/velodyne #445: Noetic PCL Build warning (-Wdeprecated-declarations)

icolwell-as, 2022-02-01:
**Please complete the following information:**
 - Ubuntu 20.04
 - Noetic
 - Installed from apt
 - Version: 1.6.1-1focal.20220107.012158

**Describe the bug**
When including `velodyne_pcl/point_types.h` for use in a ROS package, the following error is produced when compiling:

```
                 from /home/ian/autoware.ai/src/core_perception/points_downsampler/nodes/ring_filter/ring_filter.cpp:17:
/opt/ros/noetic/include/velodyne_pcl/point_types.h:66:36: warning: ‘using uint16_t = uint16_t’ is deprecated: use std::uint16_t instead of pcl::uint16_t [-Wdeprecated-declarations]
   66 |                                   (uint16_t, ring, ring)
      |                                    ^~~~~~~~
In file included from /usr/include/pcl-1.10/pcl/point_types.h:42,
                 from /opt/ros/noetic/include/velodyne_pcl/point_types.h:46,
                 from /home/ian/autoware.ai/src/core_perception/points_downsampler/nodes/ring_filter/ring_filter.cpp:20:
/usr/include/pcl-1.10/pcl/pcl_macros.h:94:9: note: declared here
   94 |   using uint16_t [[deprecated("use std::uint16_t instead of pcl::uint16_t")]] = std::uint16_t;
      |         ^~~~~~~~

```

**To Reproduce**
Steps to reproduce the behavior:

Include the header and compile it.

**Expected behavior**
No deprecation warnings.

**Additional context**
Add any other context about the problem here.

JWhitleyWork, 2022-02-01:
@icolwell-as Would you mind submitting a PR to resolve?

icolwell-as, 2022-02-02:
@JWhitleyWork, I had actually intended to do that, but then got a bit lost trying to find the real root cause. I could open a PR that would resolve it, but it might be better resolved some other way. 

Perhaps you can help clear up some of my confusion: When it comes to the version of PCL that is used with certain ROS distros (noetic for example), it is pretty much dictated by the version of PCL that is distributed from Ubuntu's repos right? in the case of Ubuntu 20.04 this appears to be 1.10.0 when I run `dpkg -l libpcl-dev`.

This version of PCL will pretty much remain the same for the lifetime of ROS Noetic right? we wouldn't expect it to get updated due to Ubuntu's "stability" approach to their releases?

JWhitleyWork, 2022-02-03:
@icolwell-as Yes, this is correct. The ROS `pcl_ros` package depends on the `rosdep` key `libpcl-all-dev`. In `rosdep`, based on [this line](https://github.com/ros/rosdistro/blob/1f553e68adcc8bdae9c16144b1de06f3460130a6/rosdep/base.yaml#L4340), this will get resolved to the Ubuntu package `libpcl-dev` in the Ubuntu distribution that you're currently using. For `focal` (20.04), that meta-package depends on 1.10.0 (see [here](https://packages.ubuntu.com/focal/libpcl-dev)). If there is ever a bugfix release, it will be updated for Focal. However, it won't ever go beyond 1.10.x.

icolwell-as, 2022-02-07:
Thanks @JWhitleyWork for confirming that with helpful links!

So with PCL 1.10.0, the following deprecation warning exists [here](https://github.com/PointCloudLibrary/pcl/blob/pcl-1.10.0/common/include/pcl/pcl_macros.h#L94). This is what produces the build warnings in my original post.

PCL 1.11+ seems to have removed those deprecation warnings, so they would no longer be an issue. 
However, since Noetic will target 1.10 for its lifetime, I think it's best to make the change here is this repo to suppress these warnings, even thought there isn't really anything wrong with the code here.

I'll open a PR shortly, thanks!

JWhitleyWork, 2022-02-07:
Makes sense to me!

closed 2022-02-07 by JWhitleyWork