ros-drivers/velodyne #447: Can't we get PointXYZIRT type of point cloud?

AtsukiOsanai, 2022-02-04:
### Please complete the following information:

- Ubuntu 20.04
- Foxy
- Installed from apt
- Version: velodyne-driver 2.1.0-1focal.20210901.194418

### Request
Now, the foxy branch only supports the point cloud type of XYZIR. I can find them in [velodyne_pointcloud/src/lib](2.1.0-1focal.20210901.194418). However, the latest master branch (yet ROS1) seems to provide XYZIRT point cloud. Do you have a plan to extend XYZIR to XYZIRT on ROS2?

vovo-4K, 2022-05-17:
Same problem for ros2-galactic

JWhitleyWork, 2022-07-13:
@AtsukiOsanai / @vovo-4K This feature was added by @kaarta-SHanna back in 2019. This is a community-developed driver (I'm just the maintainer) so I would be happy to review a PR adding this feature to the ROS2 version but, AFAIK, there is not anyone currently working on this.

closed 2022-07-13 by JWhitleyWork