ros-drivers/velodyne #455: Installation on Foxy with sudo apt

floriankittelmann, 2022-03-10:
**Please complete the following information:**
 - OS and Version: Ubuntu 20.04
 - ROS Version: Foxy
 - Built from Source or Downloaded from Official Repository: `sudo apt install ros-foxy-velodyne*`

**Describe the bug**
The packages are not available to run. `ros2 pkg list | grep velodyne` doesnt return anything. Do I need to build from source to run the velodyne node for ROS2 Foxy?

**To Reproduce**
Steps to reproduce the behavior:
1. `sudo apt install ros-foxy-velodyne*`
2. `ros2 pkg list | grep velodyne`
3. `ros2 run velodyne_pointcloud`

**Expected behavior**
`ros2 run velodyne_pointcloud` should be possible but returns that package not found

JWhitleyWork, 2022-03-12:
@floriankittelmann It looks like you might not be sourcing your Foxy install folder first. You need to make sure that you run `source /opt/ros/foxy/setup.bash` before attempting to use a `ros2` command. If this is the only version of ROS that you use, I recommend just adding `source /opt/ros/foxy/setup.bash` to the end of your `~/.bashrc` file. Please let me know if this resolves your issue.

floriankittelmann, 2022-03-12:
@JWhitleyWork Thank you for the hint! I was assuming that `. ~/ros2_foxy/ros2-linux/setup.bash` is enough to source the Foxy installation. But it seems `source /opt/ros/foxy/setup.bash` is also necessary. Unfortunately it was not described in the installation guide of Foxy https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Binary.html

closed 2022-03-12 by floriankittelmann

songanz, 2022-06-23:
Hi, I cannot find the `32e_points.launch` for ros foxy. Could you please share the file? Thank you so much!