ros-drivers/velodyne #458: compile the 2.2.0 driver failed

mujiubai, 2022-03-27:
**Please complete the following information:**
 - OS and Version: [Ubuntu 16.04]
 - ROS Version: [Kinetic]
 - Built from Source or Downloaded from Official Repository:  Downloaded from Official Repository
 - Version: [2.2.0]

**Describe the bug**
The 1.6.1 driver can compile success. But in the version, I can't get the time in the field. So I try to compile the v2.2.0 .
first, I use catkin_make compile the driver, but failed.
the error below:
`Base path: /home/enwu/catkin_ws
Source space: /home/enwu/catkin_ws/src
Build space: /home/enwu/catkin_ws/build
Devel space: /home/enwu/catkin_ws/devel
Install space: /home/enwu/catkin_ws/install
Creating symlink "/home/enwu/catkin_ws/src/CMakeLists.txt" pointing to "/opt/ros/kinetic/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /home/enwu/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/enwu/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/enwu/catkin_ws/install -G Unix Makefiles" in "/home/enwu/catkin_ws/build"
####
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/enwu/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/enwu/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - velodyne (plain cmake)
-- ~~  - velodyne_msgs (plain cmake)
-- ~~  - velodyne_driver (plain cmake)
-- ~~  - velodyne_laserscan (plain cmake)
-- ~~  - velodyne_pointcloud (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/enwu/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/enwu/catkin_ws/build/CMakeFiles/CMakeError.log".
`

**after that, I try to use the catkin_make_isolated to compile, also failed.**
`Base path: /home/enwu/catkin_ws
Source space: /home/enwu/catkin_ws/src
Build space: /home/enwu/catkin_ws/build_isolated
Devel space: /home/enwu/catkin_ws/devel_isolated
Install space: /home/enwu/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - velodyne (unknown)
~~  - velodyne_laserscan (unknown)
~~  - velodyne_msgs (unknown)
~~  - velodyne_driver (unknown)
~~  - velodyne_pointcloud (unknown)
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Error: Packages with unknown build types exist
Can not build workspace with packages of unknown build_type
`

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

JWhitleyWork, 2022-03-27:
Versions of the driver > 2.0 are for ROS2. They will not build in ROS1.

If you need the timestamp, you will have to back-port this functionality from the ROS2 version to `noetic` (the only supported ROS1 version) or switch to ROS2.

closed 2022-03-27 by JWhitleyWork

mujiubai, 2022-03-28:
> Versions of the driver > 2.0 are for ROS2. They will not build in ROS1.
> 
> If you need the timestamp, you will have to back-port this functionality from the ROS2 version to `noetic` (the only supported ROS1 version) or switch to RO
 
thank you a lot

mujiubai, 2022-03-31:
> 

how can I build the v2.0 in noetic? I have try to build in ROS2, but my file is bag file