ros-drivers/velodyne #461: 'Packet containing angle overflow, first angle: 35982 second angle: 22' constantly appears in the terminal

brandon-lawrence96, 2022-04-13:
**Please complete the following information:**
 - OS and Version: Ubuntu 20.04
 - ROS Version: Noetic
 - Downloaded from Official Repository:
 - Version: 1.6.1-1focal.20220221.110053

**Describe the bug**
When running other ros packages like the move_base navigation stack, the message 'Packet containing angle overflow, first angle: 35982 second angle: 22' constantly appears in the terminal. This message is coming from the VLP16_points.launch file. I am not sure if this will affect the navigation of my robot. 

**Additional context**
Here is a picture of what I see in the terminal. 
[image]

Rotoslider, 2022-07-26:
This has been an on going issue for years. I believe the developers do not know how to fix the problem and have given up.

JWhitleyWork, 2023-06-01:
Please test again with the latest version on `noetic` (1.7.0) and report back if this is still occuring.

tersite1, 2024-02-20:
> This has been an on going issue for years. I believe the developers do not know how to fix the problem and have given up.

lol fr? I have same issue, working on same env

tersite1, 2024-02-20:
> > This has been an on going issue for years. I believe the developers do not know how to fix the problem and have given up.
> 
> lol fr? I have same issue, working on same env

I just fixed by reinstall ROS, try 
`sudo apt install ros-noetic-desktop-full`
than try catkin_make