ros-drivers/velodyne #471: Accessing device diagnostic parameters directly through VelodynePackets

BatuhanYilmaz, 2022-08-10:
Hi all,

I know that we can access to device diagnostic parameters by forming [**HTTP GET requests**](https://velodynelidar.com/wp-content/uploads/2019/12/63-9243-Rev-E-VLP-16-User-Manual.pdf#page=79&zoom=100,152,424) through Ethernet; such as temperature, input/output voltages, motor RPM, laser state and more.

I wonder if it is possible to access device diagnostics from ```data``` bytes  of ```velodyne_msgs/VelodynePackets``` directly.

```
# Raw Velodyne LIDAR packet.

builtin_interfaces/Time stamp              # packet timestamp
uint8[1206] data        # packet contents
```

Thank you,

Batuhan

JWhitleyWork, 2022-08-10:
I don't believe so. The packets being read by the driver and turned into `VelodynePackets` messages are only the scan data, not the diagnostic data. There is a diagnostic UDP port which is assigned in the sensor config but I don't believe that the current version of the driver reads from this port or provides the data.

BatuhanYilmaz, 2022-08-11:
Thanks for the info.

Considering future driver releases, Is there any plan to add these diagnostics data into ```VelodynePackets``` as a new feature?

JWhitleyWork, 2022-08-11:
This driver is community-maintained. It does not have regular releases or a roadmap. Feel free to contribute this feature.

closed 2022-08-11 by JWhitleyWork