ros-drivers/velodyne #476: Unable to project the lidar points/ray in rviz2

SyahirMuzni, 2022-09-08:
- OS and Version: [Ubuntu 20.04]
 - ROS Version: [Galactic]
 - Downloaded from Official Repository
 - Version: 2.2.0-1focal.20220430.170126

**The bug**
Unable to project lidar points/ray in rviz2. I was able to project it at one time but without 'tf', also there was no option 'velodyne' in the Fixed Frame. So I rewrite it and success to project it. But later, I can't even project the points at all.  

**To Reproduce**
Steps to reproduce the behavior:
1. source /opt/ros/galactic/setup.bash
2. ros2 launch velodyne velodyne-all-nodes-VLP32C-launch.py
3. On a new tab, rviz2
4. Add, by topic pointcloud2
5. Fixed frame, velodyne

**Expected behavior**
I expect it to project it as usual and I want to solve the 'tf' issues afterwards.  

**Additional notes**
In the points drop down it's stated: "showing [0] points from [0] messages"

JWhitleyWork, 2022-09-30:
@SyahirMuzni This is likely a frame issue. After you run the commands in your post, can you please run the following commands in a new terminal?

```
source /opt/ros/galactic/setup.bash
ros2 topic echo /velodyne_points | grep frame_id
```

The above command assumes that your points are being produced on `/velodyne_points`. If the frame_id that is put out is `velodyne` then there is something wrong with collecting the data. If it is not `velodyne`, then use whatever `frame_id` is shown in RViz.

cjffly, 2022-11-30:
@JWhitleyWork I had the same issue as @SyahirMuzni described. After running  
```
source /opt/ros/galactic/setup.bash
ros2 topic echo /velodyne_points | grep frame_id
```
it returned the _velodyne_ as frame_id. 

Is there a way to sniff what data is passing to the velodyne driver? Since the same data collecting pipeline works for ROS1 and can be shown in rviz1 but does not work for ROS2/rviz2

JWhitleyWork, 2022-12-04:
@cjffly After finding the frame id as `/velodyne`, did you then set the Global Frame in `rviz2` to `/velodyne`?

cjffly, 2022-12-06:
> @cjffly After finding the frame id as `/velodyne`, did you then set the Global Frame in `rviz2` to `/velodyne`?

After I set the Global Frame to /velodyne, the points did show up. Thanks!

But the data shown is strange, not like what used to be in ros1. I will submit another issue related to this.

closed 2022-12-17 by JWhitleyWork