ros-drivers/velodyne #477: lidar data dosen't contain fields: name: "time" , how to get name: "time"?

JACKLiuDay, 2022-09-15:
Hi, thank you for your great. I have two velodyne16. When I run them. I found some problem. 
[image]
[image]
One has fields: name: "time" . While the other lidar does not have fields: name : "time". I want to use the lidar with fields:name: "time". Can you give me some help.

closed 2022-09-19 by JACKLiuDay

wwdlh, 2023-01-07:
你好，我遇到了和你相同的问题，请问你是怎么解决的啊

JACKLiuDay, 2023-01-10:
> 你好，我遇到了和你相同的问题，请问你是怎么解决的啊

时间有点久，我记不太清了，换下驱动试试

Hliu0313, 2023-04-26:
已解决，最新驱动包含了 time 信息。