ros-drivers/velodyne #480: how to use two vlp-16 in the ros  melodic

himhan34, 2022-10-24:
**Please complete the following information:**
 - OS and Version: ubuntu 18.04
 - ROS Version: melodic
 - Built from Source or Downloaded from Official Repository:
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]

**Describe the bug**
A clear and concise description of what the bug is.

hello. first thank you for the nice code sharing. i am also doing the same thing as you do with  two vlp 16 lidar

[image]

after following the thing that you guys done, there was no error, but i cant see mixed point clouds, 
is there any ways to see them?

[image]

my code is exactly like you.. 
<launch>

  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.2.201"/>
      <arg name="frame_id" value="vlp202"/>
      <arg name="port" value="2268"/>
    </include>
</group>

</launch>

can you help me?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'   https://github.com/ros-drivers/velodyne/issues/108

=> i followed this procedure in this repo. 
can you help me?

JWhitleyWork, 2022-10-24:
@himhan34 There are a lot of "procedures" in the issue you linked to.

- Also, you did not answer one of the questions in the template: are you building this repository from source or are you using the version from `apt`?
- What are the IP addresses of both of your sensors?
- What ports are set for the data and telemetry ports on both of your sensors?
- How are both sensors connected to your PC?

himhan34, 2022-10-24:
sorry for the small information. 

1. for this question. 
 **Also, you did not answer one of the questions in the template: are you building this repository from source or are you using the version from apt?** 

=> i installed throught git clone command, and i tried to follow the direction that people had done. 
https://github.com/ros-drivers/velodyne/issues/108

2. for this question. 
i followed the same procedure of the people , 
my ip is  
[image]

this one. 

3. for this question 
What ports are set for the data and telemetry ports on both of your sensors?
[image]

also the same as people had done.

4. for this question
**How are both sensors connected to your PC?**

now i am using jetson xavier for our lab's robot. 
and one of them is connected to the jetson directly and 
rest of them is connected with lan cable to usb cable to jetson. 

and i used this launch file , but there was no problem with catkin-make, 
after that i tried to do operate the node but the node didnt show velodyne points as well 
they was empty

<launch>

  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.2.201"/>
      <arg name="frame_id" value="vlp202"/>
      <arg name="port" value="2268"/>
    </include>
</group>

</launch>

JeongJae0815, 2022-10-26:
Hello, Can you share whole launch file you metioned?. I think I can help you.

JeongJae0815, 2022-10-26:
2,3 : 
I think you should make the configurations(gateway, host address) same on both sensors. 
Only differences is on ip address, data port and telemetry port(you set well.)

4.
In my case, I use a router to use both sensors.

himhan34, 2022-10-26:
ahh sorry for the late reply. 
i solved !

the problem was fixed frame.
to solve my problem i changed the fixed frame  in the launch file.  
and i set in the same fixed frame with 2 velodynelidars 
thanks for helping me ! 

have a nice day!

himhan34, 2022-10-28:
if you  dont mind can  i  ask  one question  about the multi  lidar
pointcloud  fusion??

JeongJae0815, 2022-11-02:
Sorry for replying late :(
How can I help you?

himhan34, 2022-11-02:
thanks for answering
now i am using  2 vlp 16  lidar

do you  know how to  change  oriemtation  of  point cloud?

JeongJae0815, 2022-11-02:
As I understand,I think you want to merge two point clouds from each sensors.
To merge two point clouds, you need homogenous matrix(positional relationship between two sensors) through calibration process(You need to search it with some keyword[velodyne, calibration]). You can merge them with the matrix in one frame.
(Please search transformation matrix)

On presentation of aligned point clouds on rviz, I don't know this driver offer the function of transformation.(cause I don't need it)
I think you can write a new ros code that publish new aligned point cloud.

himhan34, 2022-11-03:
thanks for helping, i will try!

TZECHIN6, 2022-11-08:
Hi, I have read the issue mentioned in above, and still no luck to see both lidar scan in Rviz2.

I am working on ROS2 Galactic docker environment, the build is using git and rosdep install. In my case, I can see both Lidar individually with different assigned IP (192.168.1.201:2368 & 192.168.1.202:2369). However after I modifier the launch file, there always a lidar messge saying `poll() timeout` and only one lidar scan data can be viewed.

I have checked the wireshark packages and also the rqt_graph, both of them are appeared. Below is the IP configuration:

lidar_left:

- IP: 192.168.1.201
- Dest IP: 192.168.1.100
- port: 2368 and 8308
- frame_id: velodyne

lidar_right:

- IP: 192.168.1.202
- Dest IP: 192.168.1.100
- port: 2369 and 8309
- frame_id: velodyne

One thing I would like to point out is that, even I change the frame_id to `lidar_left` and `lidar_right`, no change effects was shown. might be this is a bug...

Below is the launch file:

```
# Copyright 2019 Open Source Robotics Foundation, Inc.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions
# are met:
#
# 1. Redistributions of source code must retain the above copyright
#    notice, this list of conditions and the following disclaimer.
#
# 2. Redistributions in binary form must reproduce the above
#    copyright notice, this list of conditions and the following
#    disclaimer in the documentation and/or other materials provided
#    with the distribution.
#
# 3. Neither the name of the copyright holder nor the names of its
#    contributors may be used to endorse or promote products derived
#    from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
# "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
# LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
# FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
# COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
# INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
# BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES;
# LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
# CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
# LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
# ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.

"""Launch the velodyne driver, pointcloud, and laserscan nodes with default configuration."""

import os
import yaml

import ament_index_python.packages
import launch
import launch_ros.actions

def generate_launch_description():
    driver_share_dir = ament_index_python.packages.get_package_share_directory('velodyne_driver')
    driver_params_file_left = os.path.join(driver_share_dir, 'config', 'VLP16-velodyne_driver_node-params_left.yaml')
    driver_params_file_right = os.path.join(driver_share_dir, 'config', 'VLP16-velodyne_driver_node-params_right.yaml')
    velodyne_driver_node_left = launch_ros.actions.Node(package='velodyne_driver',
[... 44 lines cut ...]
                                                      namespace='lidar_left',
                                                      output='both',
                                                      parameters=[laserscan_params_file])

    velodyne_laserscan_node_right = launch_ros.actions.Node(package='velodyne_laserscan',
                                                      executable='velodyne_laserscan_node',
                                                      namespace='lidar_right',
                                                      output='both',
                                                      parameters=[laserscan_params_file])

    return launch.LaunchDescription([velodyne_driver_node_left,
                                     velodyne_driver_node_right,
                                     velodyne_convert_node_left,
                                     velodyne_convert_node_right,
                                     velodyne_laserscan_node_left,
                                     velodyne_laserscan_node_right

                                    #  launch.actions.RegisterEventHandler(
                                    #      event_handler=launch.event_handlers.OnProcessExit(
                                    #          target_action=velodyne_driver_node,
                                    #          on_exit=[launch.actions.EmitEvent(
                                    #              event=launch.events.Shutdown())],
                                    #      )),
                                     ])
```

himhan34, 2022-11-08:
[TZECHIN6](https://github.com/TZECHIN6)

For me, the Ros version is ros melodic. 
i think the problem might be the the destination ip, 

how about changing the destination ip different each??

himhan34, 2022-11-08:
[TZECHIN6](https://github.com/TZECHIN6) 

can you see the velodyne points each? 
if you cant see the data each (each lidar) 
i think you might set lidars again

TZECHIN6, 2022-11-08:
Just checked the start log info when running `ros2 launch velodyne velodyne-all-node....multi-launch.py`

```
[velodyne_driver_node-1] [INFO] [1667891961.376124524] [lidar_left.velodyne_driver_node]: Velodyne HDL-64E rotating at 600.000000 RPM
[velodyne_driver_node-1] [INFO] [1667891961.376217147] [lidar_left.velodyne_driver_node]: publishing 260 packets per scan
[velodyne_driver_node-2] [INFO] [1667891961.376129781] [lidar_right.velodyne_driver_node]: Velodyne HDL-64E rotating at 600.000000 RPM
[velodyne_driver_node-2] [INFO] [1667891961.376217164] [lidar_right.velodyne_driver_node]: publishing 260 packets per scan
[velodyne_driver_node-2] [INFO] [1667891961.376241653] [lidar_right.velodyne_driver_node]: Cut at specific angle feature activated. Cutting velodyne points always at 6.283185 rad.
[velodyne_driver_node-2] [INFO] [1667891961.376259142] [lidar_right.velodyne_driver_node]: expected frequency: 10.000 (Hz)
[velodyne_driver_node-1] [INFO] [1667891961.376241322] [lidar_left.velodyne_driver_node]: Cut at specific angle feature activated. Cutting velodyne points always at 6.283185 rad.
[velodyne_driver_node-1] [INFO] [1667891961.376257072] [lidar_left.velodyne_driver_node]: expected frequency: 10.000 (Hz)
[velodyne_driver_node-1] [INFO] [1667891961.376451525] [lidar_left.velodyne_driver_node]: Opening UDP socket: port 2368
[velodyne_driver_node-2] [INFO] [1667891961.376451602] [lidar_right.velodyne_driver_node]: Opening UDP socket: port 2368

```

The launch file just ignore what I have passed to them, below is the .yaml for lidar_left:

```
velodyne_driver_node:
    ros__parameters:
        device_ip: 192.168.1.201
        gps_time: false
        time_offset: 0.0
        enabled: true
        read_once: false
        read_fast: false
        repeat_delay: 0.0
        frame_id: velodyne
        model: VLP16
        rpm: 600.0
        port: 2368
        timestamp_first_packet: false
```

JeongJae0815, 2022-11-08:
I wish I could help you with this problem, but I've not used ROS2 yet.
This issue is focused on melodic system, so I think you'd better open new issue on this problem.

TZECHIN6, 2022-11-08:
However, base on that solution I further deep down to see what happen:

Here is the thing that I am not sure and hope someone could guide me:

1.  for velodyne_pointcloud config .yaml
```
lidar_left/velodyne_convert_node:
    ros__parameters:
        calibration: VLP16db.yaml
        model: VLP16
        target_frame: velodyne
        fixed_frame: velodyne
        min_range: 0.9
        max_range: 130.0
        view_direction: 0.0
        organize_cloud: true

lidar_right/velodyne_convert_node:
    ros__parameters:
        calibration: VLP16db.yaml
        model: VLP16
        target_frame: velodyne
        fixed_frame: velodyne
        min_range: 0.9
        max_range: 130.0
        view_direction: 0.0
        organize_cloud: true
```
This will introduce an error about VLP16db.yaml cannot be loaded, seem they dont recognize the new name with prefix `lidar_xxx`.

2. Base on the findings in point 1, actually, do I have to give namespace for all running node? Or only velodyne_driver_node is already enough?

JWhitleyWork, 2022-12-04:
@TZECHIN6 Please create different issues for different problems rather than continuing a long thread like this. If you are still having problems, please open a new issue.

closed 2022-12-04 by JWhitleyWork