ros-drivers/velodyne #482: how to change the orientation of the vertical setted vlp 16

himhan34, 2022-10-26:
**Please complete the following information:**
 - OS and Version: ubuntu 18.04
 - ROS Version: melodic
 - Built from Source or Downloaded from Official Repository:
 - nope
**Describe the bug**
A clear and concise description of what the bug is.

As the people had done in this git hub, i could operate the two lidars in the same, fixed frame. 
https://github.com/ros-drivers/velodyne/issues/108
unfortunately, one of my vlp-16 is horizontal and the other is vertical,

this picture is the the pointcloud from the vertical lidar,
[image]

this is a picture from the horizontal lidar 
[image]

and when  i displayed both lidars in the same rviz 
[image]

this is shown,

is there any method to change orientation of the vertical lidar.

and also my launch file is like this 

<launch>

  
  <arg name="velodyne_calib" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>

<group ns="vlp_1">

    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.201"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2368"/>
    </include>
</group>

<group ns="vlp_2">
    
    <include file="$(find velodyne_pointcloud)/launch/VLP16_points.launch">
      <arg name="calibration" value="$(arg velodyne_calib)"/>
      <arg name="device_ip" value="192.168.1.202"/>
      <arg name="frame_id" value="vlp201"/>
      <arg name="port" value="2269"/>
    </include>
</group>

</launch>

DrAndyWest, 2022-12-02:
To provide ROS with the orientation of a sensor, you would need to supply the specific transformation (in your case a simple rotation).  The [TF2 package](http://wiki.ros.org/tf2) provides this already in ROS, or you can describe the rotation as part of your robot in the [URDF](http://wiki.ros.org/urdf).

For the time being, just for testing, I would suggest using the [tf2 static_transform_publisher](http://wiki.ros.org/tf2_ros?distro=noetic#static_transform_publisher) and describing the relative position and rotation of your two lidar systems.

JWhitleyWork, 2022-12-17:
Thanks for helping, @DrAndyWest! @himhan34 Please open another issue if you have further driver-related questions.

closed 2022-12-17 by JWhitleyWork