ros-drivers/velodyne #483: Velodyne poll() timeout issue

TZECHIN6, 2022-11-03:
**Please complete the following information:**
 - OS and Version: Ubuntu 20.04
 - ROS Version: Galactic
 - Built from Source or Downloaded from Official Repository: Official Repo
 - Version: git clone -b ros2 <this repo>

**Describe the bug**
I have built a ros2 docker base image and build this velodyne_drive on top. The docker run has the argu `--network host`. I have successfully launched the node and see laser scan in Rviz2 few days ago. However, when I tried to connect to two velodyne VLP16 with a custom launch.py, this error occurs. I tried to reverse what I have did and rebuild the workspace and container, but both LiDAR give the same error now (only connect once at a time).

I can ping and login to the webUI, double checked the IP and destination is set correctly (should be). And also I can see message from Wireshark as well.

At this point, I have no idea at all where is the problem....

**To Reproduce**
Steps to reproduce the behavior:
1. Enter the docker container, create a workspace/src and git clone the repo with ros2 branch
2. rosdep install and colcon build as usual
3. source the install/local_setup.bash
4. ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py 
5. Error occur! --> Velodyne poll() timeout issue

**Expected behavior**
As it works just few days ago, I expected to start two VLP16 with the same launch file, but now, I just want to do the basic, to start a VLP16 and visualize it in Rviz2.

TZECHIN6, 2022-11-04:
```
15:22:21.256702 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.270352 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.281820 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.295079 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.319064 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.331100 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.378034 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.378467 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.378956 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.391729 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
15:22:21.403191 IP 192.168.1.201.2368 > 255.255.255.255.2368: UDP, length 1206
```

I just run tcpdump udp port 2368... even the lidar is already taken away (power off and packed in the box) there are still many traffic... is my switch flooded ?

TZECHIN6, 2022-11-04:
Guys, the problem might be caused by flooded UDP in switch. Power cycles it will solve the issue. at least in my case.

closed 2022-11-04 by TZECHIN6

ishaanamahajan, 2023-05-29:
Stuck on something similar. I can ping the lidar, but when I run `ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py` I get the poll() timeout issue. I would highly appreciate any help!