ros-drivers/velodyne #484: ros2 branch frame_id not working as expected

TZECHIN6, 2022-11-08:
First of all, this topic seem very similar to this https://github.com/ros-drivers/velodyne/pull/330

I pulled the ros2 branch as I am working in ROS2 Galactic. I am trying to run 2 VLP16 in a single launch file in the same PC with switch.

The IP and port of both LIDAR are configured and can be seen the lidar-scan individually. I follow the online setup from others, most of them changed their frame_id as lidar_left/lidar_right, which gives different name. When I change the frame_id, it keeps "velodyne" and my new setup doesnt take effect on this parameter.

I am working in a Docker environment (ROS2 Galactic ubuntu 20.04). 

Below are the questions I would like to seek help from the community:
1. Do I really have to change the telemetry port (8308) if multiple lidar are used?
2. frame_id, target_frame, fixed_frame --> what are they really do?
3. I got poll() timeout from one of the lidar when launch two lidars at the same time. I have already give them different namespace and the ip, port config are already done and tested. Therefore, I believe the root cause would be the <frame_id> at the same...

JWhitleyWork, 2022-12-04:
The answers to your questions depend on several factors:

1. What is the physical configuration of your two lidars and the switch? Are they both connected to a single port on your host PC or to two individual ports?
2. `frame_id` - this is the field in messages that gives the name of the fixed frame of reference in which the sensor data were taken. This must be different for each sensor. `target_frame` - in `rviz2`, this is the frame of reference in which the camera is placed. `fixed_frame` - in `rviz2`, this is the frame of reference to which all data are transformed if they aren't already in it.
3. This is again dependent upon your sensor configuration. Please answer 1.

Some general information about frames of reference in ROS: Each sensor must have a fixed frame of reference with it's own name (i.e. `frame_id`). Then, to visualize data from more than one sensor, you must either have a 3rd, common frame of reference (e.g. `base_link`) or pick one of the sensor's frames of reference as the `fixed_frame`. In order for `rviz2` to know how to transform sensor data from each sensor to the common reference frame or from one sensor's frame of reference to the other sensor's frame of reference, there must exist a "transform tree." A set of transformations between the frames which define their translation and orientation differences. These transforms must be published to either `/tf` or `/tf_static`. You can do this using the tool `static_transform_publisher`. If you are just using the two lidars, the tree would look like this:
```
lidar_1 <-> lidar_2
```

However, if you are using a 3rd, common frame (more likely) then the tree would look something like this:
```
base_link <-> lidar_1
  ^
  |
  v
lidar_2
```

The `<->`s indicate transforms which are published using `tf2`. I hope this helps.

JWhitleyWork, 2022-12-17:
Please re-open if there are additional issues.

closed 2022-12-17 by JWhitleyWork