ros-drivers/velodyne #486: Question: How can I convert pointcould2 to XYZIR?

horita-J, 2022-11-28:
OS and Version: ubuntu 22.04
ROS Version: ROS2 humble

I'm a beginner.
I don't know how to convert pointcloud2 to ring data in ROS2 humble.
In ROS1,  I could use the type "velodyne_pointcloud::PointXYZIR" to take ring data, however, I couldn't use it in ROS2.
Is there any methods to take ring data from pointcoud2?

I would appriciate it if  you could teach me in detail about procdures.

JWhitleyWork, 2022-12-04:
@horita-J On the `ros2` branch, the point cloud is stored in the format `PointXYZIRT` which contains fields for the X, Y, and Z position as well as Intensity, Ring, and Time. The type that you're referring to in ROS1 was based on a PCL structure but the ROS2 driver avoids using PCL to remove a very heavy dependency. For iterating over the points and getting the Ring field, I recommend taking a look at the `PointCloud2Iterator` which can be found in `sensor_msgs` [here](https://github.com/ros2/common_interfaces/blob/rolling/sensor_msgs/include/sensor_msgs/point_cloud2_iterator.hpp). Please let me know if this resolves the issue.

horita-J, 2022-12-06:
@JWhitleyWork  Thank you for your reply. I could get the ring data !!

closed 2022-12-09 by JWhitleyWork