ros-drivers/velodyne #487: Error while launching thev velodyne driver from ros2-galactic

abhilash1998, 2022-11-29:
**Please complete the following information:**
 - OS and Version: [e.g. Ubuntu 20.04]
 - ROS Version: [e.g. Galactic]
 - Built from Source or Downloaded from Official Repository: Built from source
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]

**Describe the bug**
I am using docker of  ros-galactic and after git cloning the galactic branch of the velodyne and building I get the following error
InvalidLaunchFileError: Caught exception when trying to load file of format [py]: __init__() missing 1 required keyword-only argument: 'executable'
when running ros2 launch velodyne_driver velodyne_driver_node-VLP16-launch.py 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a docker with ubuntu 20 then install ros galactic in it.
2. Create a workspace 
3. git clone --branch galactic-devel https://github.com/ros-drivers/velodyne.git
4. colcon build the workspace
5. source the workspace and run - ros2 launch velodyne_driver velodyne_driver_node-VLP16-launch.py

JWhitleyWork, 2022-12-04:
@abhilash1998 Can you please clarify? There is no launch file called `velodyne_driver_node-VLP16-launch.py` in the `galactic-devel` branch, only `velodyne_convert` or `velodyne_transform`.

TZECHIN6, 2023-04-13:
@abhilash1998 you can either manually change it or use `ros2` branch instead.

abhilash1998, 2023-04-13:
> @abhilash1998 Can you please clarify? There is no launch file called `velodyne_driver_node-VLP16-launch.py` in the `galactic-devel` branch, only `velodyne_convert` or `velodyne_transform`.

Hey @JWhitleyWork 
 This is the command
ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py

I fixed it, the fix was to convert -
“””
velodyne_driver_node = launch_ros.actions.Node(package='velodyne_driver',
                                                   node_executable='velodyne_driver_node',
                                                   output='both',
                                                   parameters=[driver_params_file]). “””

to

“””
velodyne_driver_node = launch_ros.actions.Node(package='velodyne_driver',
                                                   executable='velodyne_driver_node',
                                                   output='both',
                                                   parameters=[driver_params_file])”””

as the syntax changed from “node_executable” to “executable” in galactic.

Similar changes need to be done in the other launch files.

Thanks

JWhitleyWork, 2023-06-01:
Unfortunately (sorry for being late on this), `galactic` is now EOL. It looks like back-porting #435 would fix this issue but I could not release a new binary version. Please create a PR to back-port #435 if you would like this feature in the repo and I will approve it.

closed 2023-06-01 by JWhitleyWork