ros-drivers/velodyne #489: The message type 'velodyne_msgs/msg/VelodyneScan' is invalid

TZECHIN6, 2022-12-06:
**Please complete the following information:**
 - Ubuntu 20.04.5 LTS
 - ROS2 Galactic
 - Downloaded from Official Repository
 - ros2 branch
 - Docker (attached with docker run script)

**Describe the bug**
I have searched this error message in google... and actually there is no thread talking about this issue :(

**The real VLP16 lidar's scan data can be see in Rviz2 in real-time.** However, when running `ros2 topic hz/echo /velodyne_packets`, an error message `The message type 'velodyne_msgs/msg/VelodyneScan' is invalid` appear. I think this indicated this topic has some issues. For other published topics `/scan` `/velodyne_points` both works as usual, e.g. the message is ~10Hz in default.

As I have installed this in another PC with same OS and ROS version, it can echo/hz this topic. And I am also working on autoware project which I think this topic is important, otherwise the pointcloud cannot not be shown/used.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the docker with provided script
2. git clone this repo with ros2 branch
3. rosdep intall && colcon build as usual
4. source the ws
5. ros2 launch velodyne velodyne-all-nodes-VLP16-launch.py

**Expected behavior**
The /velodyne_packets can be echo/hz correctly.

**Additional context**
docker_run.script
```bash
#!/bin/bash
# This is the docker run script for cuda/ros2.

CONTAINER_IMAGE="sensors/ros:galactic-desktop-velodyne-v.1.0"

USER_VOLUME="$HOME/ros2_robotaxi-ws/sensors-ws/velodyne_ws:/home/ros/velodyne_ws"
USER_COMMAND=""

echo "CONTAINER_IMAGE: $CONTAINER_IMAGE"
echo "USER_VOLUME:     $USER_VOLUME"
echo "USER_COMMAND:    '$USER_COMMAND'"

# Give docker root user X11 permissions
xhost +si:localuser:root

# Run the container
docker run --name velodyne-devel -dt --restart unless-stopped --gpus all --network host \
    -e DISPLAY=$DISPLAY \
    -v /tmp/.X11-unix/:/tmp/.X11-unix \
    -v $USER_VOLUME \
    $CONTAINER_IMAGE $USER_COMMAND
```

The ros2 base image is created by myself, over 90% is reference to the official ROS2 document. If it is needed to solve this problem, I will provide it in here.

JWhitleyWork, 2022-12-07:
This usually happens when you do not source the workspace in the new terminal before running `ros2 topic ...`.

TZECHIN6, 2022-12-08:
Thx @JWhitleyWork . I close the bash terminal and source the workspace again... things just work as normal. I will close this issue. Thanks!

closed 2022-12-08 by TZECHIN6