ros-drivers/velodyne #490: Question: what does the height/width mean in header section

cjffly, 2022-12-06:
**Please complete the following information:**
 - OS and Version: Ubuntu 22.04
 - ROS Version: Humble
 - Downloaded from Official Repository:
 - Version: f149faf8392f6dee0387969bce5b518953c1fcf5

I had a working pipeline that using the velodyne_pointclouds to present data on rviz in ros1 (kinetic) environment. 

while using ros kinetic, after launch (launch/VLP-32C_points.launch), parts of the header output has
...
height: 1
width: 21360
...

Recently I migrate from ros1 to ros2 (Humble), after launch velodyne-all-nodes-VLP32C-launch.py, I found the echo of /velodyne_points returns the header contains:
...
height: 12
width: 32
...

The data feed into the velodyne driver is the same, I wonder what is the decision factor of this height and width?

cjffly, 2022-12-07:
To further help describe my issue, I recorded a short clip of what the Rviz2 shown the pointcloud.

On the left is using VeloView showing the data from VLP-32C, and it forwarded the Lidar data to a different port so the ROS2 velodyne driver can read it. On the right side is launch the node using 
```
ros2 launch velodyne velodyne-all-nodes-VLP32C-launch.py
```
and `rviz2` showing the data.

Is this a velodyne driver issue, or it is rviz configuration issue?

https://user-images.githubusercontent.com/11035640/206285111-8fe9f8b6-9f3a-4528-85be-d93979ddbc38.mp4

JWhitleyWork, 2022-12-07:
Try changing from Reliable to Best Effort. That being said, I have not seen this issue before. Is your Velodyne set for First Return, Strongest Return, or Dual Return?

cjffly, 2022-12-07:
The Lidar data is not from an physical VLP-32C, but rather simulation lidar data and modified to replica the velodyne data format. 

I Changed to Best Effort but nothing rviz still show broken points. The Velodyne is set to Strongest Return since that Factory byte is set to 0x37.

If the VeloView can render the pointcloud, I believe my simulation data is the same as a real data from VLP-32C, right?

cjffly, 2022-12-08:
I changed some rviz2 settings, the Frame rate from 30 to 1000, and the Decay Time from 0 to 0.8, it now can show at least something on the screen. Still not as good as the VeloView result though...

https://user-images.githubusercontent.com/11035640/206324925-3e0e56d9-4ca0-4cb5-9ee8-8ddb9c92e67e.mp4

cjffly, 2022-12-15:
[**Update**]
This seem to be related to parameter **cut_angle** issue. By default it is set to 2*Pi. I have to set it to a negative number to make it work, according to the instruction [here](https://github.com/ros-drivers/velodyne/tree/ros2/velodyne_driver#parameters) 

Although, I am not entirely sure how the cut_angle works, if someone could explain it, that would be very helpful.

JWhitleyWork, 2022-12-16:
Honestly, the default value for `cut_angle` should probably be a negative value to revert to the default behavior from before this feature was introduced. However, I don't believe this should cause issues with your particular setup because the feature should just "cut" the scan per ring at the given angle if it's less than the desired value. There are other things which can affect how this function works, though. I'll introduce a PR to revert that default.

JWhitleyWork, 2023-05-27:
@cjffly The `cut_angle` feature is now disabled by default in the `ros2` branch. Please confirm that this resolves your issue.

JWhitleyWork, 2023-06-01:
Closing as #497 should fix.

closed 2023-06-01 by JWhitleyWork