ros-drivers/velodyne #495: Documentation: Explaining current GPS functionality within documentation

Interaze, 2022-12-13:
Hey All, I'm looking towards integrating the velodyne ROS driver and I noticed in issue #24 it's referenced by @JWhitleyWork that GPS data is supported while IMU isn't. While I saw that was the case for gps timestamps, I wasn't able to see any information about getting the GPS data from the velodyne sensors. I was hoping to get some clarity on this and help write some documentation regarding this if I get any additional insight. 

Much appreciated

JWhitleyWork, 2022-12-13:
@Interaze Apparently I was mis-remembering. The functionality to use the *timestamp* from the GPS was implemented but actually publishing the GPS data from the Velodyne sensor was not. This is likely because the GPS data from the Garmin Puck that Velodyne recommends will not be very high quality. However, if you are interested in implementing this feature, please let me know.

JWhitleyWork, 2022-12-17:
Please re-open if you have further questions about this feature or create a PR if you want to implement it.

closed 2022-12-17 by JWhitleyWork