ros-drivers/velodyne #496: velodyne_driver output different between ROS2 and ROS1

cjffly, 2022-12-15:
**Please complete the following information:**
ROS1:
 - OS and Version: [ Ubuntu 16.04]
 - ROS Version: [Kinetic]
 - Downloaded from Official Repository:
 - Version: [1.5.2]

ROS2:
- OS and Version: Ubuntu 22.04
- ROS Version: Humble
- Downloaded from Official Repository:
- Version: https://github.com/ros-drivers/velodyne/commit/f149faf8392f6dee0387969bce5b518953c1fcf5

**Describe the bug**
Using the same pcap file, the velodyne_packets and velodyne_pointcloud output from ros1 and ros2 are different, leading the visualization in rviz show different. ros2 pipeline shows multiple reflection of the same objects at wront location. (like a ghost image)

See rviz output video below:
ROS1:

https://user-images.githubusercontent.com/11035640/207976588-6118e148-58f8-48bd-96ea-307710a9a5a3.mp4

ROS2:

https://user-images.githubusercontent.com/11035640/207976598-8ac2dbd8-e3da-4557-8ebd-91b0846aa38d.mp4

I exam the velodyne_driver output, (topic velodyne_packets), for the first output frame (using `rostopic echo -n 1` for ros1 or  `ros2 topic echo --once` in ros2), ros1 has more packets (151 packets) than ros2(128 packets). I wonder what make the velodyne driver treat the same pcap file differently?

cjffly, 2022-12-16:
**[update]**

I found this issue is has nothing to do with the driver, but related to the pointcloud calibration file. If I change all the rot_correction to 0.0, I would be normal again. 

@JWhitleyWork
Do you know how the calibration file/rot_crrection works?

JWhitleyWork, 2023-05-27:
Unfortunately I don't. Which sensor are you using and which configuration file?